openscad/openscad #2219: Unable to make the project: src/libsvg/libsvg.cc:7:10: fatal error: 'libxml/xmlreader.h' file not found

ViliusKraujutis, 2017-12-15:
I'm unable to `make` OpenSCAD. I'm getting this error:
```
src/libsvg/libsvg.cc:7:10: fatal error: 'libxml/xmlreader.h' file not found
```

[image]

You can see the console log here: 
https://gist.github.com/ViliusKraujutis/7f51abffcc855110603bcdd945b3eb46

I'm on Mac OS High Sierra.
I had OpenSCAD installed before.

t-paul, 2017-12-15:
This needs `libxml2`. The dependency build should already fetch and build that.

ViliusKraujutis, 2017-12-16:
So looking [here](https://coderwall.com/p/t7obea/installing-libxml2-for-python-in-mac-os-x), I'll be able to install it with `brew` like this:

    brew install libxml2

Right?

ViliusKraujutis, 2017-12-16:
Anyways, after installing and linking, I'm still unable to `make` OpenSCAD application on Mac:
[image]

To install `libxml2`, I've done like this:
[image]

Any ideas what's wrong now and how to resolve this issue?

thehans, 2017-12-16:
Did you try looking [here](https://github.com/openscad/openscad/blob/master/README.md) first?

There is a section called "Building for Mac OS X".  If you follow those instructions does it still fail?

kintel, 2017-12-16:
The python stuff is a red herring - we don't use any python code to interface with libxml2.

The problem seems to be that libxml2 in Homebrew is "keg only", meaning it won't be installed in the normal location. ..so we need to tell the build system where to pick it up.
As Homebrew suggests during install, you could try to set the `PKG_CONFIG_PATH` env variable to `/usr/local/opt/libxml2/lib/pkgconfig`.

It's also possible that `brew link --force libxml2` fixes this.

I haven't tried building against Homebrew in a while, so this is probably a change in newer versions of macOS and Homebrew that has gone unnoticed.

kintel, 2018-01-07:
Looking more closely at this, it uses the system libxml2, see `-I/usr/include/libxml2` in your screenshot.

..so it sounds like smth. is wrong with the base installation of your system, compiler or system libs. 
-> double check that you have this file: `/usr/include/libxml2/libxml/xmlreader.h`
..and if not, reinstall Xcode and its cmd-line tools.

closed 2019-01-25 by t-paul