openscad/openscad #2220: Two ways to crash hull() in 2D preview

atnbueno, 2017-12-16:
I have just updated to version 2017.11.12 (x64) in Windows 10 and I have found a couple of crashes when using `hull()` in 2D. The first one is when I try to use **really tiny numbers**:
```
hull() {
    translate([1, 0]) square(1e-17); // 1e-16 is OK
    translate([0, 1]) square(1e-16); 
}
```

The second one is when trying to `hull()` two **polygons with more than one point in common**:
```
hull() {
    rotate(45) square();
    scale([1/2, 1]) rotate(45) square();
}
```

I hope these simplified examples help somewhat on nailing some problem 🙂

kintel, 2017-12-16:
Also crashes Mac build 2017.12.02

kintel, 2018-01-06:
Some investigation done:

CGAL produces smth. like this in the second case:`polygon([[-0.707107,0.707107], [0,0], [0.707107,0.707107], [1.11022e-16,1.41421], [5.55112e-17,1.41421]]);`. Apparently, the two input vertices in the origin aren't exactly in the origin. This is fine, since it will just introduce another vertex really close to the origin.

However, CGAL itself thinks that the result of its own convex hull algorithm isn't strictly convex (probably because these algorithms aren't stable for double precision coordinates), and throws an exception.

A quick and dirty workaround could be to simply ignore CGAL's exception, however scary that is.
Another alternative could be to rewrite 2D convex hulls to not use CGAL.

kintel, 2018-01-16:
We could also rewrite to QHull (http://www.qhull.org/)

kintel, 2020-01-04:
Another example crash, with a quite reasonable expected real-world scenario:
```
hull() {
    translate([0, 10])
        draw_circle();
    translate([10,0])
        draw_circle();
}

module draw_circle() {
    rotate([0, 0, 358.987]) {
        rotate([0, 0, -358.987]) {
            circle(d=5, $fn=4);
        }
    }
}
```

kintel, 2024-03-27:
No longer crashes, but also doesn't result in any geometry