openscad/openscad #2222: Surface is open while it should be closed

vovcacik, 2017-12-18:
Hi,
I ran into a bug with one of my openscad model where surface is open instead of closed. Not sure about the terminology.

The model is defined by two shapes (ie. arrays of points), that are extruded and mirrorred. This constitutes a block that I stack one on each other and it creates tower. The top most block of tower that is >= 2 blocks high is broken:

[image]
*As viewed in slic3r.*

If it is broken or not depends on viewer I use:
- works in
    - OpenScad itself
    - Windows stl viewer
- is broken in
    - [slic3r][] 1.3.0-dev
    - [netfabb][] online service

Also note:

- if I introduce bit of inaccuracy it gets fixed, but I would like to avoid this
- single block is rendered fine, only stacked blocks lead to the error
- I reported this to the slic3r first - alexrj/Slic3r#4236
- is this rounding error?

#### files
[bug.zip](https://github.com/openscad/openscad/files/1567345/bug.zip)

[slic3r]: http://slic3r.org/
[netfabb]: https://service.netfabb.com/

thehans, 2017-12-18:
The stl files also look ok in meshlab and freecad for me.

MichaelAtOz, 2017-12-19:
Netfabb basic says it has holes, looks like two blocks are not joined.
[image]

I'll have a look at your code.

MichaelAtOz, 2017-12-19:
You have coincident faces.
Moving the slot a smidgeon 'embeds' one solid in the other.
>module block() {
    translate([0, -0.01, 4.65])//+ 0.001 - 0.00001

It seems to handle the mirror condition tho.

In general you can't butt two faces together, when joining, one needs to penetrate the surface of the other (by any amount). 
When differencing you also don't want to take away exactly the same face as another surface. This can make life hard, but 0.01 (depending on original size) isn't really noticeable.

vovcacik, 2017-12-19:
Ok, I see what @MichaelAtOz means; I also noticed that adding or subtracting small amounts (like 0.0001) from any axis fixes it.

If this kind of issues is to be expected I can live with it.

nophead, 2017-12-19:
It is a consequence of using floating point. CGAL can actually union surfaces that are exactly coplanar but it is hard to ensure they are when the vertices are represented by floating point numbers and there are any rotations involved.

MichaelAtOz, 2017-12-19:
> adding or subtracting small amounts (like 0.0001) 

This also helps with [z-fighting](https://en.wikipedia.org/wiki/Z-fighting) in preview. But here the size of the offset depends on the overall model size, 0.05 is a good size to use.

ofloveandhate, 2018-02-18:
as a practice, i set a small value as a parameter, and wherever i need to ensure i don't have coincident faces, i use it.  

```
eps = 0.01;
```

closed 2018-03-03 by vovcacik