openscad/openscad #2224: CGAL and rendering error on hull()

KC22C2, 2017-12-19:
Initially detected in OpenSCAD 2015.03 as included in Debian Stable. Re-tested and confirmed present in 2017.11.12 (git 7cb1e93) for Windows x86-64.
> module corner() {
>         #rotate_extrude() rotate([0,0,-10]) scale([3,10,1]) square(1);
>         *cylinder(h=10,r=3);
>         sphere(3);
> }
> 
> hull() {
>         translate([30, 0, 0]) corner();
>         translate([-30, 0, 0]) corner();
> }

The script above fails to render properly - the extruded part isn't considered in the hull() calculation, only the sphere:
[image]
Preview console also intermittently reports the following error:
> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329 

The commented-out cylinder was a debugging attempt, in case the hole in the middle of the extruded part was causing confusion. Result when the cylinder is uncommented:
[image]

t-paul, 2017-12-19:
That is basically #1081.

Using something like
```
#rotate_extrude() rotate([0,0,-10]) translate([0.01,0]) square([3,10]);
```
would prevent the single point on the rotation axis.

KC22C2, 2017-12-19:
Agreed on duplicate. Sorry for the noise.

closed 2017-12-19 by KC22C2