openscad/openscad #2227: CMake should build intermiediate files to 'objects' dir, like qmake

thehans, 2017-12-20:
Currently the cmake build generates a lot of files that end up in the top project directory, where they clutter up commands like "git status".

Here are all the files I see after running `cmake .` and `make -j8` 
```
OpenSCAD_automoc.cpp
comment_lexer.cxx
comment_lexer.hxx
comment_parser.cxx
comment_parser.hxx
lexer.cxx
lexer.hxx
moc_AboutDialog.cpp
moc_AutoUpdater.cpp
moc_Dock.cpp
moc_EventFilter.cpp
moc_FontListDialog.cpp
moc_FontListTableView.cpp
moc_LibraryInfoDialog.cpp
moc_MainWindow.cpp
moc_OpenCSGWarningDialog.cpp
moc_OpenSCADApp.cpp
moc_ParameterWidget.cpp
moc_Preferences.cpp
moc_ProgressWidget.cpp
moc_QGLView.cpp
moc_QWordSearchField.cpp
moc_WindowManager.cpp
moc_cgalworker.cpp
moc_editor.cpp
moc_groupwidget.cpp
moc_highlighter.cpp
moc_launchingscreen.cpp
moc_legacyeditor.cpp
moc_parametercheckbox.cpp
moc_parametercombobox.cpp
moc_parameterslider.cpp
moc_parameterspinbox.cpp
moc_parametertext.cpp
moc_parametervector.cpp
moc_parametervirtualwidget.cpp
moc_scintillaeditor.cpp
parser.cxx
parser.hxx
```

In comparison qmake seems to put all these intermediate files into `objects` directory, which is already .gitignore'd, so cmake would probably be best to do the same.

thehans, 2018-01-03:
I've made a commit that changes the destination of the lexer and parser generated files.  I used the `CMAKE_LIBRARY_OUTPUT_PATH` variable which seemed like the closest appropriate built in thing to set, even though the main openscad CMakeLists.txt doesn't build any libraries.  Alternatively I could change to a user-defined variable called something like LEXER_PARSER_OUTPUT_PATH for example.

I have not yet found a way to change the destination of the other `moc` generated files.  cmake has `AUTOMOC` functionality that handles this, so I looked for a way to set the destination directory for these files.

And there is a setting for configuring where the automoc generated files go:
https://cmake.org/cmake/help/v3.9/prop_tgt/AUTOGEN_BUILD_DIR.html
Unfortunately this was added in cmake 3.9, which is quite new, whereas our `CMakeLists.txt` is currently requiring minimum cmake version of 3.2.2.  (And my Linux Mint 18.3 is still on cmake 3.5.1, just as another data point)

So for cmake <3.9 it seems that these `moc_*` files will always just be placed in the `CMAKE_BINARY_DIR` and the only way to force cmake to put them in another folder would be to call cmake from that folder(an out-of-source tree build).  So you would have to do: `cd objects && cmake ..` instead of `cmake .` from the openscad project top directory.
Also then the `openscad` executable would be written to `objects`, and not the top level dir unless that is also addressed.

Another option would be to just put `moc_*` into .gitignore which seems like a kludge but not as much as running cmake from an objects dir.

Any other ideas?

kintel, 2018-01-03:
It's generally recommended to build out of source (`mkdir build; cd build; cmake ..; make -j8`, but it's nice to be well-behaved for in-source builds as well.
I would suggest implementing the cmake-3.9 solution, and the world will catch up eventually.

kintel, 2018-01-04:
@thehans Feel free to close this once you deem it solved

thehans, 2018-01-08:
Yeah I guess I'll have to get used to running cmake from a build folder.

note: I was not able to verify the cmake 3.9 setting, can only assume it works.

Anyways I suppose nothing else to do here, closing

closed 2018-01-08 by thehans

doug-moen, 2018-01-08:
@thehans said "Any other ideas?"

I use a top level Makefile which creates the build folder, chdirs into the build folder, runs cmake and then make.

So, a user of my project simply types "make", then "make install".