openscad/openscad #2229: Add "Export to polyhedron()" option

vovcacik, 2017-12-26:
Hi,
thingiverse.com has support of customizable openscad designs. Due to their [limitations][] we can't `import()` STL models however. Currently the workaround is to inline STL model to `polyhedron()` call. If we had *Export to polyhedron()* option in OpenSCAD itself this process would be lot more straightforward.

I also think it could be rather easy. You already have *OFF* export and this new functionality is very similar. Actually I created simple script that converts OFF file to SCAD script - [off2polyhedron.sh][]. Seems to work fine - tag [off2polyhedronsh][].

This will probably become irrelevant as soon as thingiverse.com removes their limitations; no ETA though.

[limitations]: http://customizer.makerbot.com/docs#limitations
[off2polyhedron.sh]: https://gist.github.com/vovcacik/e70f68cbee9cbda0565577ca3675e7f3
[off2polyhedronsh]: https://www.thingiverse.com/tag:off2polyhedronsh

t-paul, 2017-12-26:
Sounds like that would be useful only in Thingiverse, I suggest to bring that up with them.

kintel, 2017-12-26:
Would this be resolved by #1881 and #1882?

t-paul, 2017-12-26:
That would probably do as minimal solution. The user would then need to copy&paste the file content into the main scad file.

vovcacik, 2017-12-26:
That could do it. Right now it exports only the `import()` directive, e.g.:

````
group() {
	import(file = "model.stl", layer = "", origin = [0, 0], scale = 1, convexity = 1, $fn = 0, $fa = 12, $fs = 2, timestamp = 1514277611);
}
````

I suppose `import()` is not CSG node, right?

t-paul, 2017-12-26:
Yes, it is. CSG output is supposed to be a subset of "normal" scad.

The minimized version @kintel referenced would reduce that even further and substitute the `import()` with the `polyhedron()`.

vovcacik, 2017-12-26:
There is a discussion on thingiverse forum about this - [Loading STL into openscad](https://www.thingiverse.com/groups/openscad/forums/general/topic:9309). Hopefully some admin will comment on the issue.

kintel, 2017-12-26:
I agree with @t-paul that making this fully integrated sounds a bit like an application-specific workaround for missing functionality in Thingiverse, and would be better left to them.

I think a minimal version as suggested in #1881 and #1882 would be a good start. Next step could be to discuss how to improve interoperability between OpenSCAD and the CSG format.

vovcacik, 2017-12-26:
That sounds reasonable. Feel free to close as duplicate of #1881, #1882.

TLC123, 2017-12-27:
Not only useful for Thingiverse but also as a partial solution for the larger geometry feedback loop issue.

 vovcaciks script does whats needed. If the function is not deemed worthy of inclusion in the export menu i suggest its time to found a open-scad external tool pack...
#1881 kind of does that, But is it severely more complexity to have an option to export just the resulting root node polyhedron?

t-paul, 2017-12-27:
I think it's much more complex at the current state to export partial models. This is certainly something we want to have for other reasons too as that's needed for other features.

As discussed above, I'm closing as duplicate, for continued discussion, see the other tickets...

closed 2017-12-27 by t-paul

kintel, 2017-12-27:
@TLC123 The UI feature to export to polyhedrons is tracked in #1155