openscad/openscad #2230: Preview (F5) does nothing, Render (F6) works fine

vovcacik, 2017-12-26:
Not sure if I've done something wrong, but my model does not show up using the Preview function, but renders just fine. No warnings in log.

OpenSCAD 2015.03
100% reproducible
Windows 10 Pro

t-paul, 2017-12-26:
Can you please post the `Help` -> `Library Info` output.

vovcacik, 2017-12-26:
Done, see first post.

kintel, 2017-12-26:
I believe this was fixed by OpenCSG 1.4.2. Could you try a recent development snapshot and see if that improves things? http://www.openscad.org/downloads.html

vovcacik, 2017-12-26:
Unfortunately not. Behaves the same, F5 renders nothing, F6 works fine; no errors.
[Library info - dev build.zip](https://github.com/openscad/openscad/files/1587695/Library.info.-.dev.build.zip)

t-paul, 2017-12-26:
Ok, it's not a driver issue, the reason is that all the faces of the `350Clip` polyhedron have the wrong winding order. This shows up in `Thrown Together` mode as pinkish-color. If you swap any 2 values of all the `350ClipFaces` entries (e.g. `[44, 2, 129],` -> `[2, 44, 129],`) that should fix the model.

vovcacik, 2017-12-27:
It helped somewhat:

ordering | F5 | F6 | isPink | #swaps
---------|----|----|--------|-------
1-2-3 | ✘ | ✔ | ✔ | 0*
1-3-2 | ✔** | ✔ | ✘ | 1
2-1-3 | ✔** | ✔ | ✘ | 1
2-3-1 | ✘ | ✔ | ✔ | 2
3-1-2 | ✘ | ✔ | ✔ | 2
3-2-1 | ✔** | ✔ | ✘ | 1

*) original order
**) F5 preview works but there is ghosting:

[image]

t-paul, 2017-12-27:
This looks like [convexity artifacts](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?). Setting [`convexity = 3`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion#Extrude_parameters_for_all_extrusion_modes) for the polyhedron should fix the display issue.

vovcacik, 2017-12-27:
Great, that did it. Thanks for your insight and help!

vovcacik, 2017-12-27:
I feel little bad now, that I know all the answers are in [FAQ][].

[FAQ]: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ

closed 2017-12-27 by vovcacik

vovcacik, 2017-12-27:
Just for the record, I only needed to add `convexity = 3` on the polyhedron because I manually set `$fn=24` globally.