openscad/openscad #2231: linear_extrude degrades the quality of some polygons

btlvr, 2017-12-27:
I've noticed that sometimes using linear_extrude on a polygon will severely degrade the quality of the shape. 

Here is a simple example with two circles and a square, union-ed together to make a pill shape:
```
$fn = 200;

//linear_extrude(0.1)
union() {
    translate([0,-0.1,0])
        square([0.2,0.2]);
    circle(0.1);
    translate([0.2,0,0])
        circle(0.1);
}
```
When kept as a polygon, it looks perfect.
[image]

But when the `linear_extrude` line is uncommented, it looks like this:
[image]

If a single circle is extruded, it looks perfect. But if other shapes are added to the polygon, it will look bad when extruded.

Also, the value of `$fn` affects this in a strange way. If we start `$fn` small, and increase it, the quality of the shape will increase, reach a maximum, and start decreasing. 

The "sweet spot" for this particular shape seems to be at around `$fn = 60`, but it's different for different shapes. `$fn = 60` makes this shape smooth enough to look good, but with other shapes, the optimal value of `$fn$ still made the shape look bad. 

Why does this happen? Is there a workaround for it?

triffid, 2017-12-27:
[image]

Your code works fine for me. Tried a newer version of openscad?

Try playing with $fa and $fs instead - $fn should only be used when you want a specific number of faces (eg nut traps or similar geometry), not for generally setting detail level which $fa and $fs are vastly superior for.

kintel, 2017-12-27:
Closing as duplicate of #999 - should be fixed in latest development snapshot

closed 2017-12-27 by kintel