openscad/openscad #2241: Build fails to link to libpng16 on Ubuntu 17.04

sjjf, 2017-12-31:
Building from master fails for me on Ubuntu 17.04 (as of commit 656f6e871bfec8d8b69d7250e11ba871bd774667)
 because it's not adding libpng16 to the linker arguments. Adding the attached patch fixes the build for me, though I don't know if this is the correct approach (I'm not sure what's pulling in the libpng16 dependency - it may be more appropriate to move the test there). I just copied and modified the libzip.pri file - again, that may not be the best approach, but my qmake-fu isn't great.

[missing-libpng16.patch.gz](https://github.com/openscad/openscad/files/1595113/missing-libpng16.patch.gz)

kintel, 2017-12-31:
@sjjf Do you know why libpng16 is needed in the first place? I don't think we use libpng directly..

t-paul, 2017-12-31:
Please provide some more info about the actual issue. The builds on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) work fine for both Ubuntu 17.04 and Ubuntu 17.10, so I'd be a bit hesitant to add this without understanding the root cause.

sjjf, 2018-01-03:
Attached is a build log showing the failure - I really don't know enough about the whole system to be able to say what's going on, except that the final link fails with a whole bunch of undefined references to "png_foo@PNG16_0". They're specifically from libfreetype.so and libQt5Gui.so. Obviously my patch fixes that, but whether this is caused by something missing on my system or in my build script I don't know.

I've also attached the generated Makefile and the output of uni-get-dependencies.sh. I'm not seeing any other generated build files that might be informative - if you want me to I can re-run the failing build with whatever changes you need.

[openscad-build-1514959833.log.gz](https://github.com/openscad/openscad/files/1599520/openscad-build-1514959833.log.gz)
[Makefile-1514959833.gz](https://github.com/openscad/openscad/files/1599521/Makefile-1514959833.gz)
[uni-get-dependencies.log.gz](https://github.com/openscad/openscad/files/1599525/uni-get-dependencies.log.gz)

kintel, 2018-01-03:
That's odd; if Qt references libpng, qmake should pull that in without us having to manually do it.

Could you try one thing: Build any Qt example using QtGui, e.g. the most trivial MainWindow example: https://github.com/qt/qtbase/tree/5.10/examples/widgets/mainwindows/application

It would be interesting to see if that works, and correctly links with libpng.
Next step from there would be to try an figure out how our build system differs and what causes us to miss that dependency.