openscad/openscad #2246: "alternate construction" for convex faces in polyhedron missing

Parkinbot, 2018-01-02:
In older dev builds (like 2017.01) I get the following output, whenever a polyhedron is F6-rendered that contains a previously rotated convex polygon as face: 

```
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
```
In OpenSCAD 2017.11 I get a CGAL error instead: 

```
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: ss_circle.has_on(sp) File: /opt/mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h Line: 263 
```
here some test code that builds a pyramid over a rotated N-gon star: 

```
N = 6;  // gets convex if N>5
angle = 10; 

npyramid = Rx_(10, concat([[0,0,10]], star(N=N))); 

fac_top = [for(i=[0:N-1]) [0, i+1, (i+1)%N+1]]; 
fac_bot = [[for(i=[N:-1:1]) i]];
faces = concat(fac_top, fac_bot); 

polyhedron(npyramid, faces); 
cube(1); 

function star(r=10, N=6) = [for(i=[0:N-1]) 
  let(a = 360/N*i) (i%2?r:r/3)*[sin(a), cos(a), 0]]; 
  
function Rx_(w, A) = A*[[1, 0, 0], [0, cos(w), sin(w)], [0, -sin(w), cos(w)]]; 
```

kintel, 2018-01-02:
Duplicate of #2139 - fixed Dec 2nd

Parkinbot, 2018-01-05:
Just downloaded 2018.01.02 and found out, my code still doesn't execute. This issue does not seem to be fixed. Are you sure, you tested my sample code with the new version, before closing this issue?

thehans, 2018-01-05:
@Parkinbot Works for me on Linux Mint 18.3 with nightly build.  What OS are you on?

Parkinbot, 2018-01-05:
I'm using Windows 8.1 and installed the 64 bit zip version

thehans, 2018-01-05:
OK, I can confirm the bug on F6 rendering on my Windows 7 machine using the 2018.01.02 build.

kintel, 2018-01-05:
Sorry, I didn't test the snippet - made some assumptions based on the error message.

closed 2018-01-06 by kintel

Parkinbot, 2018-01-08:
Thanks. I can confirm the fix for my general sweep() code as well.

commit 3473eb1a (2018-01-06): Added a missing CGAL exception. Fixes #2246