openscad/openscad #2251: macOS Build fails on High Sierra .2

SConaway, 2018-01-05:


kintel, 2018-01-05:
Looks like a Homebrew-related issue. I haven't used the homebrew dependencies in a long while, so there could be some bitrot there..

thehans, 2018-01-05:
@SConaway is the result any different if you try the relinking command that the warning suggests?
`brew unlink qscintilla2 && brew link qscintilla2`

@kintel  It seems like using `brew bundle` is preferred way to handle installing a bunch of dependencies like this, but I have no mac to test any of it:
https://docs.travis-ci.com/user/reference/osx/#Homebrew
https://github.com/Homebrew/homebrew-bundle
At least it would avoid the "arleady installed" warnings from what I understand.

SConaway, 2018-01-06:
@thehans That did not help.

kintel, 2018-01-06:
@SConaway Do you know if setting up a High Sierra VirtualBox VM on a Sierra machine is easy? Last time I tried to VM a recent macOS relase, VirtualBox didn't support such a new OS.

SConaway, 2018-01-07:
It's not that hard to do, I'm using Parallels.

SConaway, 2018-01-07:
But VirtualBox's macOS version doesn't really matter, it still runs fine.

kintel, 2018-01-07:
Looks like qscintilla 2.10 change the name of the library from `libqscintilla2.dylib` to `libqscintilla2_qt5.dylib`, so we need to massage the build system to choose the correct name based on the combination of Qt and QScintilla versions.

SConaway, 2018-01-07:
Massage?

kintel, 2018-01-14:
This task is still looking for a developer who wants to tinker with build systems.

kintel, 2018-01-14:
FYI: The typical workaround is to build dependencies locally using `./scripts/macosx-build-dependencies.sh`.

SConaway, 2018-01-14:
Ok. Thanks.

SConaway, 2018-01-14:
Is there a way just to build `libqscintilla2.dylib`/`libqscintilla2_qt5.dylib`?

kintel, 2018-01-14:
You can pass the lib to build on the cmd-line.
That script is set up to build everything, so if you want to build only one lib, but link it against external libs, you probably want to tweak that script accordingly. You may also have to set up your environment to correctly mix and match libs from different sources (see e.g. `setenv_mac.sh`).

closed 2018-06-10 by SConaway