openscad/openscad #2257: Clarify the canonical repo for MCAD?

lostapathy, 2018-01-07:
Discovered this while working on #2256 

It's tricky to figure out which MCAD repo is supposed to be the canonical repo.  I assume it's now [openscad/MCAD](https://github.com/openscad/MCAD), but that repo shows it's forked from [SolidCode/MCAD](https://github.com/SolidCode/MCAD).  Our [README](https://github.com/openscad/openscad/blob/master/README.md) links to a [reprap page](http://reprap.org/wiki/MCAD) that links back to the [SolidCode/MCAD](https://github.com/SolidCode/MCAD) repo.

If [openscad/MCAD](https://github.com/openscad/MCAD) is current canonical source, a project owner here should contact [GitHub Support](https://github.com/contact) and they can update the "forked from" relationships so that this repo shows as the current root and resolves this confusion.

I can also follow up and update both the reprap wiki and our README to make everything match.

kintel, 2018-01-14:
Github support has been contacted.

kintel, 2018-01-14:
If you'd take on updating the docs, that would be awesome!

t-paul, 2019-07-09:
via #2265.

closed 2019-07-09 by t-paul