openscad/openscad #2259: linear_extrude with scale vector renders differently to (correct) preview

speendo, 2018-01-08:
The preview of the following code looks like expected. The rendered model seems to be broken:

    difference() {
        translate([-30,0,0]) {
            linear_extrude(height = 10,center = false, twist = 0, scale = [1,0], slices = 40) {
                polygon(
                    points=[[30, 0],[0, 10],[0, -10]],
                    paths=[[0,1,2]]
                );
            }
        }
        translate([0,0,-1]) {
            cylinder(h=12,r=1);
        }
    }

I'm quite confident that this already worked before, because I developed this code with some substantial help from the mailing list (http://forum.openscad.org/curved-surface-td8757.html) and published a model on thingiverse that uses a similar code (https://www.thingiverse.com/thing:370976).

By the way: if you comment out the `cylinder` it seems to render correctly...

MichaelAtOz, 2018-01-09:
Confirmed with OpenSCAD 2015.03-2 & 2018.01.06 (git 3473eb1) on Windows 7/64.
(you want a convexity to have F5 look good, but that is not an Issue)

kintel, 2018-01-14:
The problem is that we make certain assumptions on the convexity of objects to optimize processing.
In this case, the question is whether this object is convex or not:
`linear_extrude(height = 10, scale = [1,0], slices = 40) polygon([[30, 0],[0, 10],[0, -10]]);`

..or more generally, is there a way we can tell if a linear_extrude object is convex without evaluating it?

kintel, 2018-01-14:
Changing the line at https://github.com/openscad/openscad/blob/master/src/GeometryEvaluator.cc#L689
to `PolySet *ps = new PolySet(3, unknown);` fixes the problem, but it would be nice to catch more known-to-be-convex cases.

kintel, 2018-01-27:
I think this is accurate:
* Twist always yields concave objects
* Scale with non-square aspect ratio yields concave objects if any vertex angle is not 90 degrees

commit 6e7dfb77 (2018-01-28): linear_extrude() may yield non-convex objects when using non-uniform scale. Fixes #2259
closed 2018-01-28 by kintel