openscad/openscad #2262: Preview with difference() does not match render

matthew-viglione, 2018-01-11:
I am using a difference() to create a void in the red layer shown below. On "Preview" I get this strange crosshatching at an angle (1st image), and nothing when looking straight on (2nd image). When I render, the voids are created as expected, although they are hard to see (3rd image). In this case, the coloring available in the preview is very important to be able to see the different features and the preview isn't being correctly calculated. 

I have seen similar issues throughout the issues section which were resolved by updating OpenCSG from 1.4.0 to 1.4.2. I was at 1.4.0 when I first observed this issue, and have since updated to 1.4.2 but the issue remains. When running `./scripts/check-dependencies.sh` I do get these warnings, which lead me to believe this may be somehow related (maybe I didn't install OPenCSG correctly): 

`Warning: you have a copy of opencsg.h under /usr/local/include`
`Warning: you have a copy of libopencsg under /usr/local/lib`
`Please verify these local copies don't conflict with the system`

The code used to generate these pieces is included below, followed by my Library Info. 

[image]

[image]

[image]

     module spring(len, height, arm_width, arm_len, stub_len, c) {
       color(c){
          total_arm_width = (len - arm_len)/2;              // total width of all three parts of arm connection (stub-arm-stub)`
          stub_width = (total_arm_width - arm_width)/2;     // width of stubs  (x)

          difference() {
               // full membrane
               translate ([-len/2, -len/2, 0]) cube([len, len, height], center=false);

               // center piece
               translate ([-arm_len/2, -arm_len/2, 0]) cube([arm_len, arm_len, height], center=false);

               // left arm inner (towards square)
               translate ([-arm_len/2-stub_width, -arm_len/2, 0]) cube([stub_width, stub_len, height], center=false);

               // left arm main
               translate ([-arm_len/2-stub_width-arm_width, -arm_len/2, 0]) cube([arm_width, arm_len, height], center=false);

               // left arm outer (away from square)
               translate ([-len/2, arm_len/2 - stub_len, 0]) cube([stub_width, stub_len, height], center=false);

               // right arm inner
               translate ([arm_len/2, arm_len/2 - stub_len, 0]) cube([stub_width, stub_len, height], center=false);

               // right arm main
               translate ([arm_len/2+stub_width, -arm_len/2, 0]) cube([arm_width, arm_len, height], center=false);

               // right arm outer
               translate ([arm_len/2 + stub_width + arm_width, -arm_len/2, 0]) cube([stub_width, stub_len, height], center=false);
          }
       }
     }

OpenSCAD Version: 2018.01.10
System information: Linux 4.13.0-26-generic #29~16.04.2-Ubuntu SMP Tue Jan 9 22:00:44 UTC 2018 x86_64 Ubuntu 16.04.3 LTS 4 CPUs 7.68 GB RAM
Compiler: GCC "5.4.0 20160609" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.6.2
QScintilla version: 2.10.2
MingW build: No
GLib version: 2.48.2
libzip version: 1.0.1
Application Path: /usr/local/bin
Documents Path: /home/matthew/.local/share
Resource Path: /usr/local/share/openscad
User Library Path: /home/matthew/.local/share/OpenSCAD/libraries
User Config Path: /home/matthew/.config/OpenSCAD
Backup Path: /home/matthew/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/matthew/.local/share/OpenSCAD/libraries
[... 203 lines cut ...]
GL_KHR_debug
GL_KHR_no_error
GL_KHR_robustness
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

thehans, 2018-01-11:
This is called z-fighting and is a known behavior of OpenCSG.  The solution is that your negative shapes should not share faces with the positive shape, make them extend slightly past the face instead.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?

And color information is only displayed during preview, not CGAL.  This is intended behavior.

closed 2018-01-11 by thehans

matthew-viglione, 2018-01-11:
I did not see this when looking through the documentation. Thanks for the help!