openscad/openscad #2266: Is there any special interaction with let() and recursive functions?

NateTG, 2018-01-19:
The openscad manual contains the sentence, "[let()] can be used to store variables in recursive functions."  What does that mean?  (Is it some kind of a statement about scope, about tail recursion optimization, or something else?)

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#Functions

thehans, 2018-01-19:
I think its just using recursive functions as an example to segue into.  It can be used to stores variables in recursive and non-recursive functions.

Also see: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#let

And the special interaction is that tail optimization doesn't work with let/  see #2053

kintel, 2018-01-20:
Please use the mailing list to ask questions.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#let

closed 2018-01-20 by kintel