openscad/openscad #2276: DXF mangled into non-manifold geometry

stuartpb, 2018-01-31:
Check out https://github.com/stuartpb/tubman20-stamp, which extrudes [this DXF](https://github.com/stuartpb/tubman20-stamp/blob/master/tubman-portrait.dxf) in [tubman20-stamp.scad](https://github.com/stuartpb/tubman20-stamp/blob/master/tubman20-stamp.scad) - though the DXF appears to be fine, the resulting file has a bunch of gross overlapping polygons that make it unprintable.

t-paul, 2018-01-31:
This DXF extrudes fine for me with both release and nightly build on Linux using
```
difference() {
    linear_extrude(1) import("/tmp/jackson-silhouette.dxf");
    translate([6, 3, 0]) cylinder(r = 2, h = 10, center = true);
}
```

thehans, 2018-01-31:
Looking at one of the DXF by itself, seems that it has a bunch of self intersecting lines which I would assume are what throws off OpenSCAD.

[image]

stuartpb, 2018-01-31:
Yeah, I meant `tubman-portrait.dxf`, not `jackson-silhouette.dxf`.

stuartpb, 2018-01-31:
What's self-intersecting about the lines in tubman-portrait.dxf? They look like normal, proper holes to me: I don't see any lines overlapping.

kintel, 2024-03-27:
Status as of 2024-03-24: Works when using the Manifold backend

closed 2026-03-03 by github-actions[bot]