openscad/openscad #2278: Surface errors should reflect the file type

stuartpb, 2018-02-01:
From a file with this line:

```
  scale(raster_scale, raster_scale, h / 100) surface(file = "tubman-portrait.png", invert=true);
```

I got this error:

```
WARNING: Can't open DAT file 'C:/Users/Stuart/Documents/GitHub/tubman20-stamp\tubman-portrait.png'.
```

The actual cause of the error was that I'd made a typo in the name of the surface file (so it was trying to open a file that didn't exist), but the error message led me to believe it was because the PNG was being misinterpreted as a DAT.

thehans, 2018-02-05:
I think the solution would be to add a check before this highlighted section of code, for whether the file exists.
https://github.com/openscad/openscad/blob/master/src/surface.cc#L142-L147

As it is, it does not distinguish between "a file that is not a png" and "a file that doesn't exist".

I would write a quick fix but I'm not sure whats the preferred method for cross platform file check.

css911, 2019-03-16:
@kintel Hi... can you please assign this issue to me..

t-paul, 2019-03-16:
Ok, lets give that a go. As the comment above already mentions, it may need a little bit of staring at the current logic to come up with a nice solution. So I'd propose you share a prototype solution via PR. It may spark some discussion but please consider that a good thing.

css911, 2019-03-16:
Sure... excited to learn from the discussion as well. Thanks @t-paul.

kintel, 2019-03-16:
@css911 GitHub assignment only works for organization members, it's enough to leave a comment here that you're working on it.

closed 2019-04-06 by kintel