openscad/openscad #2284: Document that Resizing large parts of model can be very slow

oxinabox, 2018-02-05:
I wished to use the `resize` function to resize a large part of my model
which was its sized all in basically different "units" from an imported 2D image.
(That whole section was thus using the 2D images "units")

I couldn't workout why Preview was suddenly appearing to hang.
I ended up replacing it with `scale` instead, and doing it a bit more roughly than I would have liked.

Now with a bit more experience,
I believe the reason is because `resize` has run most of the CGAL render pipeline 
to get the bounding box.
Even in Preview mode.

It would be good if this was documented somewhere.
I did a few googles for "OpenSCAD resize slow" etc but didn't find anything.

related to #354

thehans, 2018-02-05:
I think it would be good to document specifically *any* operation that results in an implicit render (resize being just one of them).    I can't recall all of them off the top of my head but I know a handful of them exist.

In addition to putting a warning by the description of each such operation, it could also be helpful to have a list of these under the documentation for `render()` itself.  
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/CSG_Modelling#render

MichaelAtOz, 2018-04-29:
Done. See [wiki](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#resize).

MichaelAtOz, 2018-04-29:
Completed OP, closing to claim the bounty I posted.

closed 2018-04-29 by MichaelAtOz