openscad/openscad #2285: Most "rotate_extrude" of polygon rendering fail, on Arch Linux ?

kotchwane, 2018-02-05:
Hi,

Here is my code :

`rotate_extrude() polygon( points=[[0,0],[1,1], [1,2]] );
cube([1,1,1]); `

And here is the code of the failure : 

'ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329 '

Notes : 
1) Preview mode works perfectly ;
2) The first line alone is rendered perfectly. The problem seems to stem from the union of this first object with any other object. Plus, CGAL, if I understand, is only triggered whenr there is more than one object.

Thanks for your help,

kotchwane

t-paul, 2018-02-05:
Duplicate of #1081.

Having a single point touching the Y axis will by definition create a non-manifold object.

closed 2018-02-05 by t-paul

kotchwane, 2018-02-05:
Well, thank you ! This has been bothering me for hours...