openscad/openscad #2286: compile error undefined reference to `boost::re_detail...

pauledd, 2018-02-06:
Hi
trying to compile but fails with error:

```
 objects/cgalworker.o objects/Polygon2d-CGAL.o objects/import_nef.o objects/PlatformUtils-posix.o objects/lexer.o objects/comment_lexer.o objects/parser.o objects/comment_parser.o objects/qrc_openscad.o objects/moc_scintillaeditor.o objects/moc_ProgressWidget.o objects/moc_QGLView.o objects/moc_MainWindow.o objects/moc_OpenSCADApp.o objects/moc_WindowManager.o objects/moc_Preferences.o objects/moc_OpenCSGWarningDialog.o objects/moc_AboutDialog.o objects/moc_FontListDialog.o objects/moc_FontListTableView.o objects/moc_highlighter.o objects/moc_editor.o objects/moc_Dock.o objects/moc_AutoUpdater.o objects/moc_launchingscreen.o objects/moc_legacyeditor.o objects/moc_LibraryInfoDialog.o objects/moc_ParameterWidget.o objects/moc_parametervirtualwidget.o objects/moc_parameterspinbox.o objects/moc_parametercombobox.o objects/moc_parameterslider.o objects/moc_parametercheckbox.o objects/moc_parametertext.o objects/moc_parametervector.o objects/moc_groupwidget.o objects/moc_ignoreWheelWhenNotFocused.o objects/moc_QWordSearchField.o objects/moc_cgalworker.o   -lCGAL -lmpfr -lgmp -lopencsg -lGLEW -lboost_thread-mt -lboost_program_options-mt -lboost_filesystem-mt -lboost_system-mt -lboost_regex-mt -lglib-2.0 -lxml2 -lharfbuzz -lfontconfig -lfreetype -L/usr/lib64 -lqscintilla2_qt5 -lzip -lbz2 -lz -lQt5PrintSupport -lQt5Widgets -lQt5Gui -lQt5Concurrent -lQt5Core -lGL -lGLU -lX11 -lpthread 
objects/handle_dep.o: In function `__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > boost::re_detail_106300::re_is_set_member<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, char, boost::regex_traits<char, boost::cpp_regex_traits<char> >, unsigned int>(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::re_detail_106300::re_set_long<unsigned int> const*, boost::re_detail_106300::regex_data<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, bool)':
handle_dep.cc:(.text._ZN5boost16re_detail_10630016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb[_ZN5boost16re_detail_10630016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb]+0x1f3): undefined reference to `boost::re_detail_106300::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
handle_dep.cc:(.text._ZN5boost16re_detail_10630016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb[_ZN5boost16re_detail_10630016re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb]+0x3ba): undefined reference to `boost::re_detail_106300::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
collect2: error: ld returned 1 exit status
make: *** [Makefile:932: openscad] Error 1
```
compiling on gentoo linux with boost-1.63.0
any ideas?

kintel, 2018-02-06:
Hard to say without access to that platform, but I would try a few things:
* Can you build any stand-alone program linking against boost::regex ?
* Inspect your linker flags vs. the linker flags OpenSCAD uses; are all libraries linked in the same order with the same name?
* Inspect any differences between compile flags to `handle_dep.o` between OpenSCAD and your test program

pauledd, 2018-02-08:
Sorry, thats far beyond my knowlegde. I will ask in the gentoo forums if anyone has an idea. If I get an solution I'll be back.

t-paul, 2018-11-07:
Closing for now... if there's any more information from Gentoo we can check via a new ticket.

closed 2018-11-07 by t-paul