openscad/openscad #2292: Syntax highlighting inconsistencies

atnbueno, 2018-02-11:
I've been looking into updating the [Visual Studio Code extension for OpenSCAD](https://marketplace.visualstudio.com/items?itemName=efbenson.scad) because I've seen that some keywords were not highlighted. I took a list of keywords and tokens from the source code:
- https://github.com/openscad/openscad/blob/master/src/scadlexer.cpp
- https://github.com/openscad/openscad/blob/master/src/highlighter.cc

And then I checked the embedded editor as reference. I was surprised to see keywords like `intersection_for` and operators like `/` not highlighted either (captured below while using the *Tomorrow Night* theme):
[image]
[keywords_tokens.scad](https://www.mundoplus.tv/atnbueno/keywords_tokens.scad)

I'd love to know enough to fix this myself and submit a PR, but it's not the case :-P

MichaelPFrey, 2018-02-12:
With the operators: Try a more realistic syntax and it works:

[image]

MichaelPFrey, 2018-02-12:
scadlexer.cpp is used by scintillaeditor which is the default editor.
intersection_for can simply be added to the keyword list in scadlexer.cpp.

Question: Do you want to make the pull request? If not, I could do it.
_(Fixing part of the syntax highlighting was my first pull request here, see #2128 and I do not want to "steal" this fix from you, as we all start by picking low hanging fruits :-) )_

atnbueno, 2018-02-12:
Nice of you to offer :-) but go ahead.

MichaelPFrey, 2018-02-12:
Single-quoted strings:
**scintillaeditor.cpp**
```
		l->setColor(readColor(colors, "string", textColor), QsciLexerCPP::SingleQuotedString);
		l->setColor(readColor(colors, "string", textColor), QsciLexerCPP::DoubleQuotedString);
```
It is as simple as commenting out the SingleQuotedString line.

I was a bit confused - OpenSCAD is bypassing some logic of qscintilla for the colorScheme feature.

MichaelPFrey, 2018-02-12:
deprecated functions: The more I think about it, it would make sense to make a new keyword set, so that we can give them a separate color/style.

MichaelPFrey, 2018-02-13:
About adding a new keyword list:
I followed the trail into the rabbit hole and all I can comeback is a note:
LexCPP.cpp has only four keyword lists and even that is kind of bending their specific intended use.
Changing LexCPP.cpp is not really an option here down stream. Changing it upstream would take ages to tickle down.

I will think about adding a comment for the next person that tries to add one more keyword list.

kintel, 2018-02-13:
@MichaelPFrey There was some work done to replace the Scintilla parser with something better, done as part of a GSoC project, but it needs someone to pick up the threads and finalize it. I seem to remember that the developer (Shaina) got pretty far though: http://brlcad.org/wiki/Google_Summer_of_Code/2015#ScadLexer_for_OpenSCAD

MichaelPFrey, 2018-02-14:
Interesting, the source is in the branch [lexertl](https://github.com/openscad/openscad/tree/lexertl).
Well, that would be a bit to big for me in scope, at least at the moment.

t-paul, 2021-12-23:
Closed via #4026.

closed 2021-12-23 by t-paul