openscad/openscad #2294: undefined symbol when compiling from source

ben-horner, 2018-02-11:
At the end of the readme at https://github.com/openscad/openscad, it said to enter a new issue if I had trouble compiling from source, so here it is.  :)

I followed the instructions there, and when I tried to run the newly compiled openscad, I got this error:

```
[2018-02-11 12:32:31] [ben@beasty] [openscad (master)]
$ openscad
openscad: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0: undefined symbol: g_type_check_instance_is_fundamentally_a
```

t-paul, 2018-02-11:
That's not really a compilation issue though ;).

There seems to be an issue with finding the listed library. It's not even clear why it would try to load that as OpenSCAD uses Qt and not Gtk as user interface.

What distribution is that?
What's the output of `which -a openscad`?
What's the output of `ldd ./openscad`?

ben-horner, 2018-02-11:
It wasn't an error during compilation it's true, but after it was compiled, it didn't work.  Maybe you're saying it's a problem with my environment?  I ran the ./scripts/check-dependencies.sh and got it to say everything was OK.

Here are the answers to your questions though:

I actually checked prior to the `make install`, and which returned `/usr/bin/openscad` prior to the install, and `/usr/local/bin/openscad` after.
```
[2018-02-11 12:32:40] [ben@beasty] [openscad (master)]
$ which -a openscad
/usr/local/bin/openscad
/usr/bin/openscad
```
The `ldd` answer is big, but you asked for it.  :)
```
[2018-02-11 12:47:38] [ben@beasty] [openscad (master)]
$ ldd ./openscad
	linux-vdso.so.1 =>  (0x00007ffce6766000)
	libCGAL.so.11 => /home/ben/openscad_deps/lib/libCGAL.so.11 (0x00007fdf8bac9000)
	libmpfr.so.4 => /home/ben/openscad_deps/lib/libmpfr.so.4 (0x00007fdf8b86b000)
	libgmp.so.10 => /home/ben/openscad_deps/lib/libgmp.so.10 (0x00007fdf8b5f4000)
	libopencsg.so.1 => /home/ben/openscad_deps/lib/libopencsg.so.1 (0x00007fdf8b3d6000)
	libGLEW.so.1.9 => /home/ben/openscad_deps/lib64/libGLEW.so.1.9 (0x00007fdf8b165000)
	libboost_thread.so.1.56.0 => /home/ben/openscad_deps/lib/libboost_thread.so.1.56.0 (0x00007fdf8af3e000)
	libboost_program_options.so.1.56.0 => /home/ben/openscad_deps/lib/libboost_program_options.so.1.56.0 (0x00007fdf8acbf000)
	libboost_filesystem.so.1.56.0 => /home/ben/openscad_deps/lib/libboost_filesystem.so.1.56.0 (0x00007fdf8aaa6000)
	libboost_system.so.1.56.0 => /home/ben/openscad_deps/lib/libboost_system.so.1.56.0 (0x00007fdf8a8a2000)
	libboost_regex.so.1.56.0 => /home/ben/openscad_deps/lib/libboost_regex.so.1.56.0 (0x00007fdf8a593000)
	libglib-2.0.so.0 => /home/ben/openscad_deps/lib/libglib-2.0.so.0 (0x00007fdf8a23b000)
	libxml2.so.2 => /home/ben/openscad_deps/lib/libxml2.so.2 (0x00007fdf89edf000)
	libharfbuzz.so.0 => /home/ben/openscad_deps/lib/libharfbuzz.so.0 (0x00007fdf89c88000)
	libfreetype.so.6 => /home/ben/openscad_deps/lib/libfreetype.so.6 (0x00007fdf899ee000)
	libfontconfig.so.1 => /home/ben/openscad_deps/lib/libfontconfig.so.1 (0x00007fdf897b0000)
	libqt5scintilla2.so.12 => /usr/lib/libqt5scintilla2.so.12 (0x00007fdf89236000)
	libzip.so.4 => /usr/lib/x86_64-linux-gnu/libzip.so.4 (0x00007fdf89023000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fdf88996000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fdf8844e000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fdf87f78000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007fdf87d05000)
	libGLU.so.1 => /usr/lib/x86_64-linux-gnu/libGLU.so.1 (0x00007fdf87a96000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fdf8775c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fdf8753f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fdf871bd000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fdf86eb4000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fdf86c9e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdf868d4000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fdf8bd38000)
	libGLEW.so.1.13 => /usr/lib/x86_64-linux-gnu/libGLEW.so.1.13 (0x00007fdf86651000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fdf86449000)
	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007fdf85fe7000)
	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007fdf85c53000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fdf85a4f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fdf85835000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007fdf8beba000)
	libgobject-2.0.so.0 => /home/ben/openscad_deps/lib/libgobject-2.0.so.0 (0x00007fdf855de000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007fdf853b9000)
	libpcre16.so.3 => /usr/lib/x86_64-linux-gnu/libpcre16.so.3 (0x00007fdf85153000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fdf84f2a000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007fdf84d27000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007fdf84b24000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007fdf8491d000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007fdf8471a000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007fdf844ea000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fdf842d8000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007fdf840d5000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fdf83ecf000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fdf83ccd000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007fdf83ab4000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007fdf838af000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fdf8368d000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fdf83487000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007fdf83276000)
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007fdf817bf000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fdf815b7000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fdf813b3000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fdf811ad000)
```

t-paul, 2018-02-11:
Ok, looks like there is no direct dependency on that library. So it's probably the Qt theme loader that tries to do Gtk. So I think it's basically a Qt issue.

Some combination of `QT_STYLE_OVERRIDE` and `QT_QPA_PLATFORMTHEME` may help to switch to a different theme. Documentation for that is a bit confusing unfortunately.

ben-horner, 2018-02-11:
There did seem to be a great deal of confusion over the issue, but changing to QT_QPA_PLATFORMTHEME=qt5ct made it usable, though it looks different than before (not a problem for me).  Feel free to close the issue (or should I?), and thanks for the help!

MichaelPFrey, 2018-02-12:
Could you state your distribution including the version?

ben-horner, 2018-02-13:
I built from source, the git hash was ec8aded3241766dda3b89e26fa914d07562de429.
It was the lastest within the last couple of days.
The version says 2018.02.11
(Let me know if you need more info.)

MichaelPFrey, 2018-02-13:
I mean which distribution of Linux/Unix/Operating system you are using.

From what I can tell, most developers here are using Ubuntu in various version, so naturally things tend to work seamless on Ubuntu and most Debian based Linux distribution, while we have a blind spot for more obscure/rare Linux/Unix distributions.

ben-horner, 2018-02-13:
Ah, sorry, I don't think I'm obscure at all...
```
[2018-02-12 19:32:02] [ben@beasty] [net (master)]
$ lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial
```

MichaelPFrey, 2018-02-13:
Thank you. I still have a laptop with Ubuntu 16.04 LTS, so I will try to compile and run the current build on there. (Makes sense anyway - the laptop will end up next to the 3D printer)

t-paul, 2018-11-07:
Closing... if there's a need to follow up let us know and we can re-open (or just create a new ticket with updated information).

closed 2018-11-07 by t-paul