openscad/openscad #2296: Support for hexstr ("#C0FFEE", "#FE0") colors in color()?

atnbueno, 2018-02-12:
Would it be too difficult to add support for **HTML/CSS color strings** to `color()`?

I mean with the "#RRGGBB" and "#RGB" syntax (it cound even be extended for alpha channels ("#RRGGBBAA" and "#RGBA" ).  In regexp it would be something like `/(#([0-9a-f]{3}){1,2}/i`.

I've found so many colors in that format (rather than as 3 integers) that I think It would be a nice addition to the language.

For now what I usually do is to convert to decimal in my head and divide the vector by 255: "#C0FFEE" --> `color([192, 255, 238]/255);` :-P

t-paul, 2018-02-12:
That's already supported via #2134.

closed 2018-02-12 by t-paul

atnbueno, 2018-02-12:
Ouch. Searched (clearly not enough) and didn't check if it worked before asking. Sorry.