openscad/openscad #2299: Turning the OpenSCAD into a GUI for other scripting languages

Foadsf, 2018-02-13:
I know it might be too far fetched but I was thinking maybe OpenSCAD should follow the same path as Ipython took towards Jupyter, turning it into a language agnostic tool. The concept of code based cad design wasn't new but the style of OpenSCAD, which is kind of similar to the way LaTeX editors like TexStudio work, was revolutionary IMHO. 
There are plenty of other scripting languages out there:

- Gmesh .geo 
- NetGen .in2d
- OpenFOAM blockMeshDict
- Elmergrid .grd
- FreeCAD and OpenCASCADE python 
-  Blender Python

or even 2D graphics scripting languages:

- PostScript
- LaTeX based graphics: Tikz/PGF, asymptote, pstricks
- MetaPost
- xml/svg

CAM scripting languages: 

- G-code
- OpenSBP
- HPLG or HPGL (Hewlett-Packard Graphics Language)

for all of these the same GUI style can be used. You don't have to develop everything from the scratch. there are plenty of libraries out there for any of these. you can just wrap your GUI around them.

doug-moen, 2018-02-13:
I would be willing to participate in such a project. I think it needs to be
a separate open source project on github, with the users being projects
that are developing OpenSCAD-like CAD languages.

Foadsf, 2018-02-13:
@doug-moen great. I agree on the point being a seperate project. 
I will help too. though have very little programing/development knowledge. Do you have enough experience developing similar projects? can you take the lead?

MichaelPFrey, 2018-02-13:
You can start you own project/prototype, but I would like to say: Be careful with taking OpenSCAD as a foundation. A lot of system are tightly knotted to each other, while other parts are simply hacked together to make it work. For instance: The preferences use the same data structur as the OpenSCAD language. On the other-hand we have syntax highlighter that was written with multi-language support in mind, but customized in a way that breaks the inherit multi-language support.

Also note the slow development speed here.

What would make more sense is to write your own UI that "somehow" tabs into the other projects as "compilers", "rendering engines" and so on.

Foadsf, 2018-02-13:
@MichaelPFrey Sad to see that the development here is not exciting. 
It might be a good idea to develop to develop a new GUI, but it is also very good for OpenSCAD to separate the UI and the core compiler. There are already a lot of libraries out there for editor features like text highlighting ... which we don't have to develop ourselves. 

the final product a collaboration of multiple modules:

1. GUI: a combination of OpenGL plus windows management (e.g. WxWidgets)
2. editor features: maybe wrapping around atom or VSC
3. metta language: a language which will sit in between the existing compilers and the OpenGL

doug-moen, 2018-02-13:
I imagine this project being a C++ library based on Qt, which is a GUI
toolkit that supports Linux, macOS and Windows.

The project could be called "OpenStudio".

The 3-pane OpenSCAD style GUI would be represented by a Studio class. You
create a subclass of Studio and implement a mandatory virtual function for
recompiling the program and updating the graphics pane when the file
changes. Your main() function creates an instance of your Studio subclass
and calls Studio::run().

Foadsf, 2018-02-13:
@doug-moen Not that I'm an expert, but why not WxWidgets instead of Qt? as far as I have read it is completely open source with no vendor/proprietary bindings.
The name is also nice, how about "CAD-Studio" though?

Foadsf, 2018-02-13:
maybe if later there can also be a visual programing mode like [antimony](http://www.mattkeeter.com/projects/antimony/3/). Users develop their design in the visual mode and then select the prefered language and then export it.

t-paul, 2018-02-13:
I guess the first question would be "what's the specific goal for this feature/change". A generic text editor? A multi-language IDE? Separate output for each file? Something that combines multiple languages in a single project?

GUI and core are separate, there's just no main focus to maintain a separate compilation framework, but with the slow move to `cmake` that might be even simpler than with the still official `qmake` build. Also there are not much text editor components that integrate with Qt. Pretty much the only active one we found when trying to switch to a more feature rich component was QScintilla.

I'd support the changes to make the UI a bit more generic, but right now that would need additional help. That could even be a [GSoC](https://summerofcode.withgoogle.com/) project if we can find a motivated student interested in working on that topic. The benefit would be to make use of the existing infrastructure for building installers and packages for various platforms and distributions. I understand that new projects sound always simple and easy and getting some results over a weekend is fun. Maintaining a project over a long time is a slightly bigger challenge, but can be fun too.

Foadsf, 2018-02-13:
@t-paul the goal would be to do what Ipython team di by turning it into Jupyter and including other scripting languages like R and Julia. Not everybody knows OpenSCAD and not everybody needs/wants to learn the language. But if it becomes a language agnostic code based (aka generative indirect design) platform, it would help a lot of people. For example consider the OpenFOAM or Elmer users. they have some options:

1. scripting their model in the specific languages embedded in these softwares. without having any visual feedback.
2. creating the geometry in a separate software like Gmsh or NEtGen through GUI and then editing the script separately to achieve the parametric model they want

It would be great if they could just have a OpenSCAD or [antimony](http://www.mattkeeter.com/projects/antimony/3/) to write their code and see it in the same place.

Foadsf, 2018-02-13:
@mkeeter I thought maybe you might be interested in this discussion.

Foadsf, 2018-02-13:
I have posted this also [here on reddit](https://www.reddit.com/r/openscad/comments/7x9o6w/openscad_to_take_a_same_path_as_ipython_towards/?st=jdlwh8uf&sh=435f8ce7).

doug-moen, 2018-02-13:
Qt is GPL 3, which is an open source licence. It's the most full featured
and popular GUI toolkit, and there's a big ecosystem of plugins. There's a
lot of development muscle behind the project. So, from that point of view,
it seems a reasonable choice. OpenSCAD uses Qt, and uses the Qscintilla
editor plugin.

I was thinking about forking an existing OpenSCAD-like GUI, like OpenSCAD
itself, or maybe libfive Studio, which looks like cleaner code, both of
which use Qt.

In the end, the choice will be made by whoever is writing the code, whoever
is the lead developer.

To get this project off the ground, it would be good to have more than one
C++ developer, and more than one project that uses the library. Otherwise,
it risks turning into a one-person project.

Foadsf, 2018-02-13:
@doug-moen 

1. You have for sure more experience than I. and GPL3 is great. SoI think Qt it is. I just had some concerns about former Nokia and propiaratry issues... 
2. I had not seen [libfive Studio](https://libfive.com/studio/) but it seems amazing. maybe we can invite them also here?
3. This [Guile](https://www.gnu.org/software/guile/) is being mentioned twice by now. it is used to develop libfive Studio as well as [Ao](https://www.mattkeeter.com/projects/ao/) as was pointed to me in this [reddit comment](https://www.reddit.com/r/openscad/comments/7x9o6w/openscad_to_take_a_same_path_as_ipython_towards/du6oyu5/?st=jdlzz71q&sh=435f4779). Ao is also developed by Matt Keeter the same person who developed antimony coincidentally! he seems like the right guy to be involved in this.
4. regarding the development part I will be there but as a mechanical engineer with little C and python experience there is a lot for me to learn yet.

> more than one project that uses the library
this I did not understand. would you please explain?

doug-moen, 2018-02-13:
@Foadsf said:
>> more than one project that uses the library

> this I did not understand. would you please explain?

I am working on a project called Curv: it's a 3D modelling language that kind of competes with OpenSCAD and libfive. My GUI is very weak right now, and I need to get a better one. I would rather use a separate open source project to provide my GUI, than write my own. And that's why I am interested in this proposed project. I would be using this proposed GUI library as a dependency for my Curv project.

However, it's not good if I am the only person who is uses the new GUI library.

I don't know if the OpenSCAD team is interested in splitting their code into two separate libraries, a core library, and a generic reusable GUI library. They might not be interested in maintaining a new public API. So this discussion is probably not appropriate as an OpenSCAD issue.

However, there may be people developing their own language that might be interested in using this GUI library, who might be interested in joining this project. And they probably aren't reading this issue, so again, this is not the best place for this discussion.

And yes, Matt Keeter (@mkeeter) is awesome. Libfive is a great project.

kintel, 2018-02-13:
Our code is pretty much split into GUI vs. non-GUI, but without a clean interface. There is bound to be application specific bits in such a GUI, so I'm not convinced that the integration into a new GUI will happen against that API surface, but rather some sort of adapter API between the core and the GUI APIs. 
Also, as @MichaelPFrey pointed out, please don't look at OpenSCAD for a good example of how to write a Qt GUI. Our UI code it quite the mess.

Foadsf, 2018-02-13:
I had also seen [this awesome demo from Bret Victor](https://vimeo.com/66085662). It is not exactly what I had in mind when posting this but still inspiring.

doug-moen, 2018-02-13:
@Foadsf Re: Bret Victor.
Check out https://TheBookOfShaders.com/. Find one of the code examples. Click on one of the numeric constants in the code, and a slider pops up. Use the slider to dynamically vary the constant. Notice the smooth animation in the graphics pane beside the source code. This is a Brett Victor inspired feature that I would like to have in the Curv GUI.

mkeeter, 2018-02-14:
@doug-moen That interactivity in the Book of Shaders is really cool!

@Foadsf No need to email me, tweet at me, tag me in this issue, *and* open up an Antimony issue to get my attention.

I appreciate the idea here, but I don't see anything actionable.  There's a strong path-dependence in how these tools evolved, and often tight coupling from the UI to the data model (Antimony's graph engine is particularly bad on this front).  Studio is better, but that's primarily because the GUI is simpler.  Uncoupling is always possible, but there are only so many hours in the day, and most of the people in this thread are working on these projects in their spare time...

If you want to make progress on this, a proof-of-concept is worth a thousand ideas.  You could start with Studio (since it's a pretty minimal text-editor + OpenGL window), fork it, strongly decouple the GUI from the `libfive` backend, then work with Doug to plug Curv in as an alternate backend.

Another philosophical inspiration could be the [language server protocol](https://github.com/Microsoft/language-server-protocol), which tries to turn the `editors * languages` combinatorial explosion into a more sensible `editors + languages` sum, by defining a common protocol for editors and languages to target.  Making a working proof of concept would be a good forcing function to figure out what the common protocol needs to contain.

Foadsf, 2018-02-14:
@mkeeter sorry for spamming you :blush: I do really wanted to draw your attention and happy I succeed. My goal is to encourage and facilitate the use of indirect code based (aka generative) CAD. My motivations for supporting code based CAD are:

- version control: without proper version control (e.g. GIT) collaboration and development is meaningless and the proprietary platforms like SolidWorks require huge investments for their proprietary PDM/PLM
- parametric design: Although parametric design already exist in almost all of the commercial CAD platforms but their solvers can't handle all of the complex relations defined through the GUI. They all break at some point. Code based CAD in combination to a graphical programing interface like antimony can solve this issue IMHO.
- Open Source hardware: There have been many attempts towards open source hardware projects but they haven't been as successful as softwares. One of the main issues in my opinion is the fact that incremental development does not exist. referring to GIT again!
- Object Oriented Programing: everytime I want to design a parametric gear in SolidWorks it grinds my gears! I feel like killing myself :). There are already attempts in the OpenSCAD and FreeCAD world for standard libraries. encouraging the code based CAd will boost the development of parametric standard hardware libraries as well. Just imagine you could have a class for gears and then specify the type of gear and the set of parameters you want to couple with your main design and then the object is created from the proper class automatically. 

@doug-moen Lets choose a name for the project and create separated github repo for it. My proposal is some form of CAD or generative arts plus the term studio as you suggested.

Foadsf, 2018-02-14:
A very similar and interesting project is the [CadQuery GUI](https://github.com/jmwright/cadquery-gui) which used Electron for the GUI design.

doug-moen, 2018-02-14:
Good find. Note that neither OpenSCAD nor Curv can be fully implemented in
a web browser, due to missing APIs in HTML/Javascript. So Electron doesn't
work for me. Email me at doug@moens.org to get this discussion out of the
OpenSCAD issues tracker.

Foadsf, 2018-02-15:
in this comment I want to make a list of all similar GUI projects I have found for future reference:

Name | language | licence | GUI toolkit | OS 
------------ | -------------|------------ | ------------- | -------------
OpenSCAD | OpenSCAD | GPLv2 | Qt | Windows, Linux, Mac, BSD
[libfive ](https://libfive.com/)([Ao](https://www.mattkeeter.com/projects/ao/)) |  Guile Scheme | LGPL2, GPL2 | Qt | Windows, Linux, Mac
[Antimony ](https://github.com/mkeeter/antimony)| Visual | MIT | Qt? | Linux, Mac
[kokopelli ](https://github.com/mkeeter/kokopelli)| Python | MIT | Qt? | Linux, Mac
[cadquery-gui](https://github.com/jmwright/cadquery-gui) | CadQuery | Apache2 | Electron | All
[Cirkuit ](https://wwwu.uni-klu.ac.at/magostin/cirkuit.html)| Gnuplot, TikZ, Circuit Macros | ? | Qt? | ?
TikzEdt| PGF/TikZ | ? | ? | Windows
ktikzQtikZ  | PGF/TikZ | ? | KDE4/Qt | Linux
[GCLC/WinGCLC](http://poincare.matf.bg.ac.rs/~janicic//gclc/) | ? | ? | ? | Windows, Linux
[Ghostscript Studio](https://archive.codeplex.com/?p=ghostscriptstudio) | Ghostscript | ? | ? | ?
[MEPer ](http://cseweb.ucsd.edu/~s1pan/MEPer/)| MetaPost  | ? | ? | ?
[SVG-edit](https://github.com/SVG-Edit/svgedit) | SVG | ? | ? | web
[Gnuplot Editor](https://sourceforge.net/projects/gpedit/) | Gnuplot | ? | Qt | All

mkeeter, 2018-02-15:
Minor feedback:
- `libfive` / Studio *doesn't* yet work on Windows (the core library compiles, but Guile does not, so the UI can't be built)
- Antimony *does* work on Windows, through MSYS2 (documented [here](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#windows-experimental))

Foadsf, 2018-02-15:
@mkeeter If I may ask, what is the main advantage of Guile over lets say Python? Are there any reason(s) you decided to go with it instead?

Foadsf, 2018-02-15:
For those of you who might be following this discussion. I have made [the GCADstudio organization](https://github.com/GCADstudio) , **G**enerative **CAD** **studio**, to follow up the discussion.

doug-moen, 2018-02-15:
A key strength of OpenSCAD is that it is a functional/declarative language,
which supports a really nice style of programming. Python is an
imperative/scripting language, and Python has very poor support for writing
code in a functional/declarative style. But Guile supports this style
extremely well. For the same reason, Curv is a pure functional language.

mkeeter, 2018-02-15:
- [Sandboxing](https://www.gnu.org/software/guile/manual/html_node/Sandboxed-Evaluation.html), so it's safe to execute unknown scripts
- Since Scheme is homoiconic, it's easy to walk the AST and do rewriting operations (like when you drag shapes and variables change, [first video](http://libfive.com/studio/))
- Lots of other fun Scheme tricks, like the [`sequence` macro](https://github.com/libfive/libfive/blob/master/doc/guide.md#the-sequence-macro) and defining custom syntax for vectors and variables
- It's designed for embedding, and I had a ton of trouble deploying embedded Python (mixed results here; it's a pretty heavyweight Scheme for embedding).

kintel, 2018-02-24:
Closing this as this it's no longer an OpenSCAD issue

closed 2018-02-24 by kintel

thomastjeffery, 2020-07-04:
Where did this discussion go?

I feel like I just read the first half of a conversation here. Did everyone regroup somewhere else and finish it, or did you all just go your separate ways and forget?

doug-moen, 2020-07-04:
Nothing ever came of this. It's a big project, since it has to satisfy the requirements of, and integrate with, multiple 3D modelling languages. More work than building a GUI for a single language. And, if the only thing that GCADStudio offered was a simple 3 pane GUI (text editor/text console/graphics pane), then it wouldn't be compelling enough for individual 3D modeller projects to pay the cost of integration. There would have to be a lot more juice to get people excited.

The most exciting framework of this nature that I'm aware of is [Glamorous Toolkit](https://gtoolkit.com/). The problem is that it doesn't support 3D graphics.