openscad/openscad #2301: Polyhedron + minkowski -> SIGSEGV

LubosD, 2018-02-13:
This code makes OpenSCAD crash:

```
$fn=25;

module smoother() {
    polyhedron(points = [
        [100, 5, 0], [70, 7.5, 0], [0, 7.5, 0],
        [0, -7.5, 0], [70, -7.5, 0], [100, -5, 0],
        [100, 5, 1], [70, 7.5, 5], [0, 7.5, 5],
        [0, -7.5, 5], [70, -7.5, 5], [100, -5, 1],
    ],
    faces = [
        [0, 1, 7, 6], [1, 2, 8, 7],
        [2, 3, 9, 8],
        [5, 0, 6, 11],
        [0, 1, 2, 3, 4, 5],
        [6, 7, 10, 11], [7, 8, 9, 10],
        [5, 4, 10, 11], [4, 3, 9, 10],
    ]);
}

minkowski() {
    smoother();
    sphere(r=0.5);
}
```

Reproduced under Debian and Gentoo. Crash occurs in `CGAL::polygon_mesh_to_nef_3`.

kintel, 2018-02-13:
Confirmed on macOS using OpenSCAD-2017.12.23

thehans, 2018-02-13:
Just quickly tested this on the nightly Linux build on Linux Mint 18.3.  

I do see the following error when previewing, but it does not crash:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe != 0 File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 212 
```

The nightly build is built against CGAL 4.7.  
I suspect @LubosD version is built against newer 4.9 or 4.10?  Can you post your Help -> Library Info

LubosD, 2018-02-13:
@thehans CGAL 4.10.

kintel, 2018-02-13:
It crashed for me using 4.11

kintel, 2018-02-16:
It's definitely due to the polygons not being in order; when passing this to CGAL, we end up with two unclosed objects which causes the `CGAL::Polyhedron_3()` constructor to crash at https://github.com/openscad/openscad/blob/master/src/cgalutils.cc#L82

We have a check for closed & valid meshes in place just before that line, which we could easily enforce. Just not sure if that breaks other marginal use-cases. We'll have to add a testcase or two to test for this before enforcing that constraint.

PR #5229 (2024-07-21): Don't attempt Nef3 conversion of bad Polyhedron to avoid crash
Fixes #2301: Makes #2301 work in Manifold mode. It will still fail (but not crash) in CGAL mode
closed 2024-07-21 by kintel