openscad/openscad #2302: ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1106

wmy367, 2018-02-16:
### Console ### 
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1106 
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 2 minutes, 5 seconds
Rendering finished.
### Openscad File ###
```
module grap_circle(r=100,n=10) {
    for(i=[0:(floor(360/n)-1)]){
        rotate(360/n*i)translate([r,0,0]) children(0);
    }
}

module base(height=1000,n0=10,n1=10,t0=90,t1=-360){
    linear_extrude(height=height,twist=t0,slices=100)grap_circle(r=height/5,n=n0)translate([-height/5,0])square([height/5*2,height/200],center=true);
    linear_extrude(height=height,twist=t1,slices=100)grap_circle(r=height/5,n=n1)translate([-height/5,0])square([height/5*2,height/500],center=true);
}

base(height=100,n0=24,n1=24);
```

### When execute render,it throw error ###

MichaelPFrey, 2018-02-16:
The for loop looks off to me:
```
module grap_circle(r=100,n=10) {
    for(i=[0:(floor(360/n)-1)]){
        rotate(360/n*i)translate([r,0,0]) children(0);
    }
}
```

Now I just add an echo to see the actual values:

```
module grap_circle(r=100,n=10) {
    for(i=[0:(floor(360/n)-1)]){
        rotate(360/n*i)translate([r,0,0]) children(0);
        echo(concat(i,  360/n*i));
    }
}
```

and the result is:
```
ECHO: [0, 0]
ECHO: [1, 180]
ECHO: [2, 360]
ECHO: [3, 540]
ECHO: [4, 720]
ECHO: [5, 900]
ECHO: [6, 1080]
ECHO: [7, 1260]
ECHO: [8, 1440]
ECHO: [9, 1620]
ECHO: [10, 1800]
```
I would rewrite that as
```
module grap_circle(r=100,n=10) {
    for(i=[0:(n-1)]){
        rotate(360/n*i)translate([r,0,0]) children(0);
        echo(concat(i,  360/n*i));
    }
}
```
Doesn´t fix your problem tough

kintel, 2018-02-17:
It's because you're twisting a very thin rectangle with a sharp angle, causing the result to be self-intersecting.
You could try two things:
1. Instead of using a `square()`, use a polygon with more vertices, allowing the twist to be build more smoothly
2. Increase the slices to make the result not self-intersect.

wmy367, 2018-04-09:
:-)
Thinks a lot!

closed 2018-04-09 by wmy367