openscad/openscad #2303: High memory consumption for module recursion

thehans, 2018-02-18:
I have a test function that creates nothing but a deeply nested bunch of group() nodes.
Moderate levels of recursion result in exploding memory usage.

```
module recur(i) {
  if (i > 0) recur(i-1);
}

recur(400);

// Memory usage for various i values, using F5 preview
// 0    29.0MB
// 100  34.2MB
// 200  78.5MB
// 300  190.2MB
// 400  442.6MB
// 500  1.0GB
// 600  1.3GB
// 700  2.0GB
// 800  2.9GB
// 900  4.0GB
// 1000 5.4GB
// Each tested from a freshly opened instance of openscad
```
Memory usage seems to increase quadratically with the depth of recursion.

I tried to do some profiling and came up with this, which appears to show that most of the time is being spent in NodeDumper::visit

[image]

I don't know the Node traversing code well, but should it even be calling NodeDumper for a basic F5 preview?

kintel, 2018-02-18:
Thanks for looking into this.
NodeDumper is typically used to generate cache keys for the module cache.

thehans, 2018-02-18:
OK I thought maybe NodeDumper was only meant for exporting .csg file.

It seems like *some* cache here is not being utilized efficiently/correctly, but I still don't know in what way.

I tried some more profiling.  This is including the recent string building optimizations, and I lowered the recursion depth in the scad, and increased perf record_size to max because I think the stack frames were getting truncated in the previous screenshot.  In this flame graph you can see the stack all the way to the bottom for most of the samples:
[image]

So in mainwin.cc:121 there is the call to Tree::getString which is what kicks off the whole NodeDumper traversal, and it has this FIXME comment:
```
			// Dump the tree (to initialize caches).
			// FIXME: We shouldn't really need to do this explicitly..
			this->tree.getString(*this->root_node);
```

Any insights?

thehans, 2018-02-18:
I still barely understand what I'm doing with `perf` here, and ideally would be recording something more specific to memory like malloc calls or something.

kintel, 2018-02-18:
You should be able to to remove that line in `mainwin.cc`. This cache is used to generate cache keys for the GeometryCache, and that cache will simply call the same function one it gets to evaluating geometry, see e.g. https://github.com/openscad/openscad/blob/master/src/GeometryEvaluator.cc#L46

..but there is no reason why we cannot do this lazily.

kintel, 2018-02-18:
In terms of memory usage, my guess would be deep stacks with way too many `Context` objects in it.
You could try smth. like this:
* Modify the .scad to add an `else` to the termination clause.
* Pull out gdb, set a breakpoint in the else handler, e.g. https://github.com/openscad/openscad/blob/master/src/control.cc#L327
* Inspect the stack at this point. See if you can identify a lot of similar stack frames and figure out what memory is being allocated in those stack frames. You could even increase memory footprint of certain classes and see if that correlates to your memory usage increases.

thehans, 2018-02-18:
I tried removing the line from `mainwin.cc` and now the F5 preview is instant with no excessive memory usage.  

However, if you "Display CSG Tree...", then the memory usage shows up again.  

Also now that F5 is efficient I tried F6, and that still very inefficient, and much slower than was F5 before any of the optimizations.

Here is a profile running F6 now:
[image]

On the right side, it looks like its spending about 1/3rd of cpu time just updating the progress bar, is it spamming the update too much?

kintel, 2018-02-18:
The first thing you describe kind of makes sense. The preview doesn't do geometry evaluation, so it won't use the cache. In F6 mode, it will have to hit the cache and we'll amortize the nodedumper cost across the rendering pass.
However, this indicates that it's the nodedumper itself that's the problem, which puzzles me..

kintel, 2018-02-18:
About the progress bar; we should only ever call that a 1000 times: https://github.com/openscad/openscad/blob/master/src/mainwin.cc#L738

Perhaps you found a bug somewhere there?

kintel, 2018-02-18:
Another thing you could try: Run this all on the cmd-line to factor out any GUI-related issues.

thehans, 2018-02-18:
About the NodeDumper efficiency, am I right in understanding that each cached node is holding a string representing a subtree of the .csg file?
If these csg strings are being used internally for module cache keys, can we strip indentation tabs in this case?  Is that the whole issue here?  Because high nesting results n^2 amount of tab characters.

kintel, 2018-02-18:
We already do that: https://github.com/openscad/openscad/blob/master/src/Tree.cc#L48

thehans, 2018-02-18:
The problem is NodeDumper maintaines its own cache though, which isn't stripped.  And stripping it in the Tree.cc with that regex is where its spending all its execution time, which can be seen in the most recent profiler screenshot.

As a quick test I removed the parts in NodeDumper that write `currindent` to the stream:
here https://github.com/openscad/openscad/blob/master/src/nodedumper.cc#L49
and here: https://github.com/openscad/openscad/blob/master/src/nodedumper.cc#L84

With that change, I can run up to `recur(1650)` and the F6 results are instantaneous, and memory consumption doesn't go above 40MB or so.
Higher than 1650 gives an error of ```ERROR: Recursion detected calling module 'recur' ```,  but that's a separate issue.

So this will break all the dumptests, but it seems like the proper solution would be to have some mechanism to specify whether the NodeDumper is working for a .csg export vs internal module cache. 
And the .csg export should just build the whole thing once and ignore NodeDumper's own cache.

kintel, 2018-02-18:
Nice catch!
Yes, there is not inherent need for the .csg export and nodedumper to reuse the same content.
It would be nice to not have to duplicate too much code though, so perhaps the easiest would be to instrument this in `NodeDumper`'s constructor (use cache, use indentation etc.) ?

thehans, 2018-02-18:
I'm trying to figure out a solution that can play nice with both .csg export and Tree::getString

The way in which NodeDumper traverse the tree is really hard for my brain to follow, and is tightly integrated with cache usage such that I can't figure out a good way around it.

I'm wondering if instead of trying to completely avoid the cache when including indentation, to keep using it as general "hit" checking, but fill it with dummy values of empty string or something?

thehans, 2018-02-19:
Also is this code doing anything?  They don't seem to be referenced again?
https://github.com/openscad/openscad/blob/master/src/openscad.cc#L973-L975

kintel, 2018-02-19:
The NodeDumper is just a visitor. I can see how it's hard to follow; I tried to balance the cost/benefit of adding node nodes vs. adding new operations, but I couldn't work around the need for the visitor pattern infrastructure to get the needed double dispatch functionality.
I'll have another look and see if I can suggest a clean way around it.

kintel, 2018-02-19:
OK, I see what you mean now. A non-caching version of the NodeDumper would be a lot simpler to implement (we wouldn't need the whole `visitedchildren` construct, it could stream directly to a stream etc.). Perhaps the best way forward to to separate this out into two visitors, and look at code duplication later one, when the current problems are solved.

kintel, 2018-02-19:
I think the stack objects in `openscad.cc` can be killed.

thehans, 2018-02-19:
Just a mention that the ProgressBar issue probably was a red herring.  That profile was done at a low recursion level and total execution time was only 3s or so, making the widget draw a significant portion of the time.  I guess its reasonable for it to take a second or so to redraw a widget 1000 times?  I haven't noticed any trend of ProgressBar time increasing with larger processing loads.

thehans, 2018-02-19:
So the duplication in NodeDumper nodecache is still bothering me.  We should try to avoid storing duplicated substrings at every level.  I think the way to do this is to dump to a single stream from the root node, convert to a string, and the nodecache would store just the start,end index used to get the substring from that root string.

thehans, 2018-02-19:
It could use `stringstream.tellp` during traversal to record the start and end of each node as it is built.

kintel, 2018-02-19:
Right, so instead of building this bottom-up, we build the entire string once and index it.
I like that approach!

thehans, 2019-12-23:
I think this has been addressed for a while now, not sure why I left it open.  Closing.

closed 2019-12-23 by thehans