openscad/openscad #2307: Costumizer: replacing .scad with .json

MichaelPFrey, 2018-02-18:
While reading the code of the customizer, I found this:
`this->jsonFile = scadFile.replace(".scad", ".json").toStdString();`
which has the implication, that when I have a file called "this.scad" with a path like this:
`/home/userName/myProjects/files.scad/this.scad`
the json file ends up in the folder
`/home/userName/myProjects/files.json/this.json`
That is actually a cool hidden feature.

But thinking deeper, this has to be a bug/issue: Maybe someone is using a username like students.scad or mu.scadet, resulting in OpenSCAD trying to write in someone elses userprofile...

So, this has to be a bug. If no one disagrees, I would write a fix for that issue.
Most robust would be to remove the file ending from the scad file and then to add .json.
(Keeping in mind, that the file ending can be any capitalization of .sCAd, none, ...)

If someone thinks that this is a feature, we need a very clear specification to workaround issues like  usernames randomly containing .scad.

kintel, 2018-02-18:
I agree. The intention was to place a file next to the `.scad` file.

MichaelPFrey, 2018-02-18:
[Boost replace_extension](http://www.boost.org/doc/libs/1_53_0/libs/filesystem/doc/reference.html#path-replace_extension) sound perfect for this issue.

ParameterWidget.cc has two instances of the replace in question. readFile and writeFileIfNotEmpty.
I wonder why writeFileIfNotEmpty is neither using nor updating this->jsonFile.
Anyway: I am working on it, the pull request will contain various changes/fixes regarding Costumizer File Handling.

PR #2311 (2018-02-23): costumizer: improved file handling
* move the writeable check from "readParameterSet" to "readFile"
   * This also fixes an issue with empty (0 bytes) json files: Empty json files cause an exception, causing readParameterSet to return false, without actually checking if the file would be writable.
* checking if the file exists, before trying to check for readability or writeabilty
*central setFile function with replace_extension instead of replace (closes #2307)
closed 2018-02-23 by kintel