openscad/openscad #2309: Inefficient bracket operator on string values

thehans, 2018-02-19:
I've been doing some tests on my [L_system.scad](https://gist.github.com/thehans/a1494db8046a58832e2ebb10a5908a66) script, which generates fractal curves and features very heavy string processing.

Here is a profile of running `dragon_curve(n=15, tree=false);` from the above script:
[image]

A large chunk of the time is spent in `g_utf8_strlen` from glib.
This comes from class `bracket_visitor`'s `Value operator()(const std::string &str, const double &idx) const` 
https://github.com/openscad/openscad/blob/master/src/value.cc#L697
```c++
  Value operator()(const std::string &str, const double &idx) const {
    Value v;

    const auto i = convert_to_uint32(idx);
    if (i < str.size()) {
			//Ensure character (not byte) index is inside the character/glyph array
			if (i < static_cast<unsigned int>(g_utf8_strlen(str.c_str(), str.size())))	{
				gchar utf8_of_cp[6] = ""; //A buffer for a single unicode character to be copied into
				auto ptr = g_utf8_offset_to_pointer(str.c_str(), i);
				if (ptr) {
					g_utf8_strncpy(utf8_of_cp, ptr, 1);
				}
				v = std::string(utf8_of_cp);
			}
    }
    return v;
  }
```

So every time we access a character in a string, it has to iterate over the entire string to determine its length in utf8 characters.
I would like to propose that we define some sort of string wrapping class that ensures `g_utf8_strlen` is called only once per string, and caches the result.

I have some ~~really bad code that doesn't compile~~ "psuedocode" that hopefully at least gives the general idea of what I'm picturing:
```c++
class str_utf8_wrapper : std::string 
{
public:
	const glong& get_utf8_strlen() const {
		if (cached_len == boost::none) {
			cached_len = g_utf8_strlen(this->c_str(), this->size());
		}
		return *cached_len;
	};
private:
	mutable boost::optional<const glong> cached_len;
};
```

We would also need to ensure that the string its wrapping is immutable, so the cached size is always correct.

thehans, 2018-02-19:
The proposed wrapper class would then be used to replace `std::string` in the following line:
https://github.com/openscad/openscad/blob/master/src/value.h#L202
```c++
  typedef boost::variant< boost::blank, bool, double, std::string, VectorType, RangeType > Variant;
```

doug-moen, 2018-02-19:
Seems good. One thing: I would instead use 'glong' to declare cached_len.
I'd initialize cached_len to glong(-1). It's faster and consumes half the
memory of boost::optional.

doug-moen, 2018-02-19:
You could represent a string as an array of utf-32, which is 32 bits per
code point. Then len(str) and str[i] would be O(1).

It might suck for huge strings, but then the user is probably being
clobbered by the cost of string concatenation. Which can be mitigated by
using the "Rope" data structure, which is basically a binary tree of string
segments, with a length on each tree node. In that case, there is less
problem with using utf-8. I'm not saying that Ropes are worthwhile for
OpenSCAD: it's not really a string processing language.

New programming languages are treating strings as arrays of grapheme
clusters, not as arrays of code points. For example, a lot of newer emoji
characters consist of multiple code points. This could be an issue if you
are iterating through a string and expect each code point to correspond to
a character, as interpreted by the `text` primitive. In Swift, a string is
an array of grapheme clusters. So the Swift equivalents of len(str) and
str[i] are O(N), and that can't be fixed due to the nature of Unicode, and
due to their choice to represent strings as arrays.

I think the best approach is to represent a string as a Rope, and define a
character as a Unicode grapheme cluster. Each node in the rope contains the
number of grapheme clusters in that subtree.

And this is why Curv strings are restricted to ASCII. Unicode is a
nightmare.

PS, a Unicode string is not a good choice for a data structure, see
discussion above. Strings should just hold human readable text, ie
utterances in a natural language. You generally shouldn't be indexing into
them. For L-systems, maybe try using lists instead. Maybe a list of short
strings.

thehans, 2018-02-19:
I've got a basic length cache implementation going.  There's still another `g_utf8_strlen` call in the `builtin_length` module, because I think changing this requires changing all of `tostring_visitor` to return the new `str_utf8_wrapper`.

Made a profile from PR #2310 codebase with this fix, and I can't even find `g_utf8_strlen` in the graph now!
[image]

thehans, 2018-03-27:
Well, now that time spent in `g_utf8_strlen` has been minimized, its only more obvious how much time could be saved by avoiding `g_utf8_offset_to_pointer`.  This would be possible by converting to a fixed-size encoding as @kintel has suggested.

Looks like the function to use for this would be `g_utf8_to_ucs4_fast` (or `g_utf8_to_ucs4` if we want error checking):  https://developer.gnome.org/glib/stable/glib-Unicode-Manipulation.html#g-utf8-to-ucs4

I could try doing the conversion by using the same wrapper class concept, and caching the ucs4 (equivalent to UTF32 as i understand) converted string as I did with caching the utf8 string length.

closed 2026-03-04 by github-actions[bot]