openscad/openscad #2314: [Feature Request] Symbol for echo statement

oysteinkrog, 2018-02-20:
I very often find myself adding echo statements to debug my code.
It would be nice to be able to use "#" or some other symbol to print variables :)

kintel, 2018-02-20:
Could you clarify? Are you looking for some sort of `printf()`-style string formatting functionality?

thehans, 2018-02-20:
Sounds like he means adding a new type of modifier character that works on expressions and cause them to be automatically echo'd?

oysteinkrog, 2018-02-21:
Yes, I meant what thehans describes :)

t-paul, 2018-02-21:
I'd vote for not implementing this.
* We already spent a number of symbols for other cases but it's all one class of markers (for modules/geometry), adding additional cases will make things more obscure
* It's not obvious which scope the print have, e.g. for `#x + 3`
* If typing echo is too much, it almost can be implemented like that using `echo()`, just not with special characters

```
function D(x) = echo(x) x;

function blah(a, b) = a + D(b);

c = blah(3, 6);

// ECHO: 6
```

oysteinkrog, 2018-02-21:
1. I think it's a good argument that it can make the language more obscure/difficult to understand. One would have to balance that against the additional benefit it provides. Obviously I think the benefit is greater than the cost/downside here :P
2. I actually find it to be quite obvious in that specific case, but I see your point that there may be other examples where it is not so obvious. This is however solved quite easily by simply documenting the behavior, and perhaps poll the existing users to see what people would intuitively expect.
3. That's true, but that seems even more difficult to understand to me (I mean for other people reading the code). That example is actually a great argument for having a standardized echo operator.

t-paul, 2018-02-21:
That `echo()` above is the standard operator. This is already implemented and that code works in the dev snapshots.

MichaelAtOz, 2018-02-22:
I'm with @t-paul on this. Tho I'd use `function e(n) = echo(n) n;` but that's just me...

kintel, 2018-02-22:
I tend to agree with @t-paul; even with such a syntax, we'd need to define scoping rules, probably by introducing parentheses, which would reduce this feature to a single-character function invocation.

We could, of course, take a few dozen steps back and look if there really are similarities between such a feature and the debug/highlight operators. If we could generalize the mechanisms for these operators, that could be helpful in general as they tend to be treated as exceptions rather than rules inside the OpenSCAD parser and evaluator.

This is a larger discussion. If anyone is interested in thinking deeply about this and take a stab at how this would fit cleanly into OpenSCAD, I'd support the endeavor.

closed 2026-05-06 by jordanbrown0