openscad/openscad #2316: Customizer improvement

MichaelPFrey, 2018-02-23:
I want to use this issue as todo list to work against.

UI Improvement:
- [ ] on start up of openscad, various buttons of the customizer have no function, but are enabled
   - #2311 should be merged, before starting on this
   - [ ] disable the buttons on/before start up (e.g. the UI file) and/or disable the whole widget?
   - [ ] connect functionality as soon as possible (e.g. on saving the scad file, getting the first parameters, ...)
- [ ] translation
   - [ ] check if all relevant strings are translatable
      - #2295, #2315
   - [ ] update the translation files
   - [ ] fully translate for at least one language (ultimately proving that the translation really works)
- [ ] descriptionLoD: add an option to hide the variable name
- [ ] implement a way to rename parametersets
- [ ] if json file exists and is writeable, but is not readable ask user before overwriting the json-file

Code Structure:
- [ ] multiple inheritance

New features:
- [ ] groups that are only shown under certain conditions
- [ ] variables that are only shown under certain conditions

Documentation: (https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP#Customizer)
- [ ] make sure that every feature has an example
   - [ ] /* [Tab Name] [some more] */ is missing 
- [ ]  explain, where the customizer is looking for variables / where it stop collection variables

Compatibility:
- [ ] https://www.thingiverse.com/
   - [ ]  http://customizer.makerbot.com/docs

see also #1781 and #1751  but not that those todo list might not be uptodate.

see also https://github.com/openscad/openscad/issues?q=is%3Aissue+is%3Aopen+label%3ACustomizer

MichaelPFrey, 2018-02-24:
descriptionLoD:  to hide the variable name
[image]
if json file exists and is writeable, but is not readable, ask user before overwriting the json-file, showing the file name
[image]

kintel, 2018-02-24:
In terms of #1840 - passing the editor content may fix that. I haven't really looked at the details..

MichaelPFrey, 2018-02-25:
Parameters from includes are more bugged then I initially thought.

collectParameters only gets the text of the main file and the line number of the assignment (without knowing in which file the assignment was). The result is, that it might use an unrelated main comment to describe the variable of an include.

That beeing said: We should write a test case for that, although I am not yet certain about the exact passing criteria.

t-paul, 2018-03-24:
I think we should skip special variables, e.g. `$fn` when saving parameters, latest master seems to pick it up from the `Hidden` section in https://www.thingiverse.com/thing:421886

MichaelPFrey, 2018-03-25:
I think we should simply skip variables in the hidden section. Saving those values in the json file is a bug, regardless if the variable is a normal or special variable.

Based on the discussion on the mailing list, one use case for the customizer could be for a designer to have different views/cuts and levels of quality for preview/designing and actual printing. (The bare SCAD file beeing the printable version and the "customized" version for designing and preview)

An other use for the special variables could be to get a low-poly look model. (using the special variables as artistic tool)

Of course, a designer can always
```
myvariable = 10;
$fn = myvariable;
```
So, skipping special variables does not mean making them inaccessible from the customizer.

Still: Saving any variable of the hidden section is a bug, while whether or not to include the special variable in the customizer is a preference thing (which the designer of a model can always overwrite).

MichaelAtOz, 2018-04-14:
This [Thingi](https://www.thingiverse.com/thing:739573) has lots of controls for testing. But only the usual suspects...

In so doing, with 2018.04.01, I noticed that with Automatic Preview UN-checked, changing the drop-down next to it, ie Show Details etc, causes a preview. I would not expect that.

tsingi, 2018-04-14:
Lots of good font manipulation in there, thanks Michael.

thehans, 2018-05-05:
found a test case which gives a critical error on the nightly builds:
`test = 0; // [[]]`

There should probably be a bunch of tests for all the different combinations of nested tokens and weird punctuation, etc. that users could put into customizer drop-down box item labels or strings without thinking.  I experienced a few seemingly random crashes using the nightly, while doing a lot of modification to customizer params in the scad. But I couldn't reproduce any other crash after creating a debug build from master, except the nested bracket example above.

thehans, 2020-06-04:
@MichaelPFrey Thank you for all the work you've put into this.  Your original post looks incredibly detailed and comprehensive.

I see that many of the checkboxes have PRs noted next to them which are now merged.  Would that indicate that those can be checked off?
If you have some time, could you go over it once more and verify which items can be checked off?
It may even be helpful to close this and start a new one with only the remaining issues copied over, for easier digestion.

closed 2026-03-05 by github-actions[bot]