openscad/openscad #2317: Editor vs. Cache

ednisley, 2018-02-23:
Commit #2313 seems to have broken the Editor's cache update operation.

A [simple bracket with engraved text](https://gist.github.com/ednisley/177978fc52c012b0134f0f7ab6373ea5) shows the problem.

I edit the source in a separate editor (Kate), change the "MPCNC" string in the **text()** function to something else, then save the file. OpenSCAD reloads the source and runs the usual recompile-preview operation, but the model view doesn't change to match the new string value shown in the editor pane.

Manually Previewing (F5) the reloaded source has no effect.

Clearing the cache, _then_ manually Previewing (F5), correctly updates the model view.

It's not a problem with **text()**, because changing the numeric value of, say, **WallThick** doesn't change the size of the upright slab. However, the slab _does_ move as the code recomputes its position based on the new thickness.

It's not an Auto-reload problem, because manually changing the source code in the Editor pane also doesn't update the previewed model.

I'm unqualified to dig into the commit. Perhaps an Auto-reloaded / edited source file doesn't trigger the new hashing operation, so the Auto-Preview grabs the previously cached geometry?

Thanks ...

t-paul, 2018-02-23:
Is there a way to capture this issue in a test case?

ednisley, 2018-02-23:
Every so often I do **git pull / make** to keep up with the new shiny. You folks have a great track record of not breaking things and, if I can help a bit with simple gotchas, it's a bonus.

I fat-fingered the ID selection list, which presented #2313 as the second item down. I mistakenly thought it was the next-most-recent commit.

Soooo: a175d3a is broken, 01f36e7 works. I got there by git bisecting from early January, as it's been a while since I did an update.

kintel, 2018-02-23:
Testing the cache has been a bit annoying since most of our tests are black-box regression tests.
There is a workflow described here for testing one of the caches manually: https://github.com/openscad/openscad/blob/master/testdata/modulecache-tests/README.txt

kintel, 2018-02-24:
@ednisley Please try the current master (i.e. after #2313 was merged)

ednisley, 2018-02-25:
Returned to master: fixed!

Thanks ...

closed 2018-02-25 by kintel