openscad/openscad #2332: Apt download on Bionic

Lomomn, 2018-03-06:
There is no release on Ubuntu Bionic 18.04.

t-paul, 2018-03-06:
We don't have control over packages in Ubuntu. If that means it will ship without OpenSCAD, that's sad but not much we can do (pinging @chrysn ... any ideas?).

There will be nightly builds on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) once 18.04 is released.

thehans, 2018-03-06:
This site is a good reference, and confirms that openscad not in bionic:
https://packages.ubuntu.com/search?keywords=openscad&searchon=names&suite=all&section=all

However openscad-mcad exists, which depends on openscad, so that's obviously an error.
https://packages.ubuntu.com/bionic/openscad-mcad

This issue needs to be reported in the appropriate Ubuntu issue tracker though.

t-paul, 2018-03-06:
Problem is likely that the package is [currently not in Debian testing](https://packages.debian.org/search?keywords=openscad&searchon=names&suite=all&section=all). So that might be very bad timing.

Ghostkeeper, 2018-05-07:
Link to the Ubuntu bug report is here: https://bugs.launchpad.net/ubuntu/+source/openscad/+bug/1766555

They mention a workaround there: Install the Debian/Sid package. See that link for instructions.

t-paul, 2018-05-09:
Thanks for the link. Hopefully there's a way to get back in. Especially as the workaround can break pretty much any time in case libraries change.