openscad/openscad #2336: Generating an APPCRASH when exiting, closing or quitting OpenSCAD

theoneandonlyjonasv, 2018-03-09:
Everytime I exit OpenSCAD it generates the error window "openscad.exe has stoped working".  I can still open my files, preview, compile, etc...I can still get back into OpenSCAD and working in new or previous files.

Problem details:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	openscad.exe
  Application Version:	0.0.0.0
  Application Timestamp:	564d2b59
  Fault Module Name:	ig4icd64.dll
  Fault Module Version:	8.15.10.2869
  Fault Module Timestamp:	506dfa58
  Exception Code:	c0000005
  Exception Offset:	000000000036c3d2
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	5dce
  Additional Information 2:	5dce2c9d09f6e6a96476308c6e493951
  Additional Information 3:	8b2b
  Additional Information 4:	8b2b36bfbe6d34a22cd7bab141e58c71

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt

t-paul, 2018-03-09:
That looks like a crash in the intel graphics driver. Maybe have a look if there's a new version available.

Otherwise, please add the information asked for in the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?)

theoneandonlyjonasv, 2018-03-12:
I doubt there's a newer version but I'll definitely give that a shot.  Thanks t-paul.

kintel, 2024-03-29:
I assume this is fixed by now. @theoneandonlyjonasv if you have any updates, please let us know

kintel, 2024-12-22:
Closing as fixed due to no further reports

closed 2024-12-22 by kintel