openscad/openscad #2337: Prefereces/Features - clicking checkbox causes Compile/Preview

MichaelAtOz, 2018-03-10:
2018.03.10-64
Click on any of the feature checkboxes, console shows compile happens.
I presume that isn't meant to happen?

MichaelAtOz, 2018-03-10:
Note: Checkboxes on the Editor & Advanced tabs do NOT cause a compile.

MichaelPFrey, 2018-03-11:
#2185 is unrelated.

MichaelPFrey, 2018-03-11:
I can confirm the behavior described in the issue for both Windows and Linux.
The behavior is not new (I was going back to the december 2017 windows build and was able to reproduce it).

Now the question is: Is this a bug or a feature?
First of all, it does not affect release builds, as the (experimental) features are all hidden and disabled in release builds.
We are just used to use experimential builds and features here on github.

It could be a feature, as enabling an experimental feature can be required to run (or test) some openscad-files, as they may require experimental commands. Particuallarly echo expression is one of those that someone might want to turn on and off quickly for a simple compatibility check.

MichaelAtOz, 2018-03-11:
I vote Bug.
While preview is usually fast, I have some models which are not, also some with render() which can be very slow the first time.
I would not want a compile to come out of the blue.
In terms of testing experimental features, pressing F5 is not that hard?

kintel, 2018-03-12:
The only thing I can think about would be that "Automatic reload and compile" may want this sort of automation, but even then it's a bit surprising behavior. Especially considering that some previews may be time consuming (e.g. minkowski sums)

MichaelPFrey, 2018-03-13:
here is the offending line:
`connect(Preferences::inst(), SIGNAL(ExperimentalChanged()), this, SLOT(changeParameterWidget()));`
and then
```
void MainWindow::changeParameterWidget()
{
	if (Feature::ExperimentalCustomizer.is_enabled()) {
		viewActionHideParameters->setVisible(true);
	}
	else {
		viewActionHideParameters->setChecked(true);
		hideParameters();
		viewActionHideParameters->setVisible(false);
	}
	emit actionRenderPreview();
}
```
Well, that confirms bug.

MichaelAtOz, 2018-04-29:
Confirmed fixed in 2018.04.01 thanks.

closed 2018-04-29 by kintel