openscad/openscad #2342: Error Memory exhausted

edwardramg, 2018-03-14:
Good morning,

I have a problem with an Open Scad file of 250,000 lines of code:
translate([0.00000000000000, 0.00000000000000, 1])
cube([0.01,0.01,0.01]);
translate([0.00000000000000, 0.02004008016032, 1])
cube([0.01,0.01,0.01]);

... Until 250,000 lines:
translate([10.00000000000000, 10.00000000000000, 1])
cube([0.01,0.01,0.01]);

When I open it with OpenScad, I get the following error:
ERROR: Parser error in line 39982: memory exhausted 
ERROR: Compilation failed! 

Any ideas on how to solve it?

Thank you,

Edward

thehans, 2018-03-14:
What version of OpenSCAD are you running?

Can you attach a a single .scad file that reproduces the problem?

I did a quick google search and found this entry for "memory exhausted": https://www.gnu.org/software/bison/manual/html_node/Memory-Exhausted.html
which links to
https://www.gnu.org/software/bison/manual/html_node/Recursion.html#Recursion
With the takeaway being:
> Any kind of sequence can be defined using either left recursion or right recursion, but you should always use left recursion, because it can parse a sequence of any number of elements with bounded stack space.

I wonder if there are any cases of right recursion in openscad parser code that can be changed to left recursion as the bison docs recommend?
I haven't looked very deep at how the parser works yet.

edwardramg, 2018-03-14:
> Enviado el: miércoles, 14 de marzo de 2018 02:19 p. m.
>
> If you really want to display hundreds of thousands of cubes like this I would recommend putting the translation points into one large list, and using a for loop over the list to translate, that would at least cut down the file size somewhat.
>
> points = [
>
>  [0,0,1],[1,0,1], //... etc.
>
> ];
>
> for (p = points) translate(p)
>
>   cube([...]);

Thank you, I'll try your proposal and I'll let you know.

Saludos cordiales,

Atte... 

!Mario Eduardo Ramos García

edwardramg, 2018-03-15:
Thank you for your help, the code of my scad file is already working properly, with the code suggestion you gave me.

Best regards

Atte... 

!Mario Eduardo Ramos García

thehans, 2018-03-16:
@edwardramg I'm glad that suggestion helped.  May I ask what your end goal output is for this?  Are you just using OpenSCAD to visualize the cubes (presumably generated from some other script or program)?  Are you exporting to anoter file format?  I would imagine attempting to do any CSG operations on the group of cubes for example would take prohibitively long to compute.

Note: This issue looks like a duplicate of #1991 which I searched and found after this issue was opened.  However I could not reproduce the test case for that and ran into dependency issues.  Having a single example file (or at least a self contained archive) really helps, so thanks for that.

Although your code is working I think there's still some room for improvement to the parser and CSG tree code in OpenSCAD, which I'm looking into.  This issue gives me something to test against.

ccorcos, 2018-09-22:
I'm having a similar issue. Trying to render a giant geodesic sphere. I generated the code from another program because the math is pretty complicated. Any options to boost memory capacity?

[geosphere.txt](https://github.com/openscad/openscad/files/2408222/geosphere.txt)

Ayowel, 2019-02-18:
This is a bit old, but in case you still have the problem @ccorcos .
Taking a look at your file, you could perform the same change thehans recommended. Put the main operation in a for loop to improve your memory usage at compile time (and make use of openscad loops optimisations):

```openscad
for (data = [ /* your data */ ]) {
       // using the data structure : [[x,y,z],[colors], [translate factors]]
	x = data[0][0];
	y = data[0][1];
	z = data[0][2];
	length = norm([x,y,z]);
	b = acos(z/length);
	c = atan2(y,x);

	color( c = data[1] )
	translate([data[2][0]*scale,data[2][1]*scale, data[2][2]*scale])
	rotate([0, b, c])
	cylinder(h = length*scale, r1 = 1*thickness, r2 = 1*thickness, center = false);
}
```

I generated the array's content by applying the following search/replace pair on your current code:
```regex
x[0-9]+ = (-?[0-9.e\-]+);\ny[0-9]+ = (-?[0-9.e\-]+);\nz[0-9]+ = (-?[0-9.e\-]+);\nlength[0-9]+ = norm\(\[[xyz0-9,]+\]\);\nb[0-9]+ = acos\([a-z0-9/]+\);\nc[0-9]+ = atan2\(y[0-9]+,x[0-9]+\);\n\ncolor\( c = \[([ 0-9.,]+)\] \)\ntranslate\(\[(-?[0-9.]+)\*scale,(-?[0-9.]+)\*scale, (-?[0-9.]+)\*scale\]\)\nrotate\(\[[ 0-9., bc]+\]\)\ncylinder\(h = length[0-9.]+\*scale, r1 = 1\*thickness, r2 = 1\*thickness, center = false\);\n
```
```regex
[[\1, \2, \3],[\4], [\5, \6, \7]],
```
This was enough to get rid of the 'memory exhausted' error for me. Please le me know if it is not enough.

thehans, 2019-11-01:
A workaround was originally given for @edwardmg, but eventually the root of the issue was fixed by #2343 

@ccorcos It was a while ago since you commented, but if this issue is still relevant to you, please test on a recent build of OpenSCAD and it should be no problem.

Closing.

closed 2019-11-01 by thehans

zchlrnr, 2021-02-14:
@thehans terribly sorry if this is a faux pa, but I'm having this exact issue using OpenSCAD 2021.01

Here's the model that's doing it. It says it exhausts parser memory at line 29980.

[Sol400Model.txt](https://github.com/openscad/openscad/files/5978635/Sol400Model.txt)

jordanbrown0, 2021-02-14:
As Hans says, restructure your data to have a list of points and a small loop that renders them.  That's a much simpler structure from the POV of the OpenSCAD language system.

The result looks like:
```
points = [
    [0.442500,0.350000,0.000000],
    [0.415407,0.284593,0.000000],
[...]
    [5.650000,3.937500,-0.550000],
    [5.650000,0.062500,-0.550000],
];

for (p = points) {
    translate(p) sphere(r=0.07);
}
```

and runs fine in my OpenSCAD where the original failed.

thehans, 2021-02-14:
Thanks @jordanbrown0, simplifying the code definitely makes it look nicer, more concise, and easier on the parser... but there is also a simple one-line fix in the parser (a case of left recursion which got overlooked the first time), so we can avoid this variation of the error in the future, regardless of verbose code.

commit 2e5f9e33 (2021-02-15): Use left recursion for "inner_input" in parser.  Fixes #2342 (again)
commit ddaa68ec (2021-02-15): Merge pull request #3671 from thehans/issue2342

Use left recursion for "inner_input" in parser.  Fixes #2342 (again)