openscad/openscad #2344: [Feature Request] Text geometry information in user land

NateTG, 2018-03-17:
I've been tinkering with approaches to the request for 'beveled text' on the forum, and I think that it's possible to do something in userland that's going to be faster than the double minkowski method that was described in the thread.  (It would involve implementing 'solid skeleton' algorithm to generate the bevels, and a tesselation algorithm so that the result can be 'rendered' with polyhedron().)

[roof.scad.txt](https://github.com/openscad/openscad/files/1821248/roof.scad.txt)

http://forum.openscad.org/Chamfered-3D-text-td23162.html

The problem is that in order to be useful, this requires access to the font geometry in userland.  Now, I can scrape geometry using post script for development, but that's not a particularly accessible thing for most people.

[roof.scad.txt](https://github.com/openscad/openscad/files/1821232/roof.scad.txt)

Access to text geometry would also make for a much easier and better way to do 'text on cylinder' and 'text on sphere' stuff.

Since font geometry is not being generated by the geometry engine, this seems like it should be pretty feasible.

MichaelAtOz, 2018-03-17:
+1. I was looking at this, but got lost on the minutia.
IIRC, the glyph geometry should be readily available, but how two+ glyphs pack together may be harder (see kerning & hiting in the link below - that's where I got lost), it also differs in vertical/horizontal directions etc.
For me, just the info on a glyph would be enough, such as:
(see https://www.freetype.org/freetype2/docs/tutorial/step2.html)
[image]
I did this a while back trying to get a handle in the metrics, IIRC (??) it is in the terms used by freetype and/or other metrics which are needed to know where actual geometry starts/ends:
[image]
I suspect two bounding boxes may be enough, the box of the glyph symbol, the box of the glyph with spacing.

MichaelAtOz, 2018-03-17:
Ohh, I see we are talking about two different bits. I want sizing, Nate wants geometry...

NateTG, 2018-03-17:
> Ohh, I see we are talking about two different bits. I want sizing, Nate wants geometry...
--

Well, they are intimately related.  The stuff you're asking for is probably more useful for sane people.

kintel, 2018-03-17:
For the general purpose counterpart to this request, see #2076

kintel, 2018-03-17:
Some aspects of #1881 could overlap with this (or provide a temporary workaround)

jordanbrown0, 2023-11-18:
Text metrics (sizing information) were added in PR #3684.  Actual outlines are available in PR #4478 if it ever merges.