openscad/openscad #2345: macOS fullscreen preferences does not create a modal dialog

ze, 2018-03-18:
Opening the preferences window in openSCAD doesn't create a modal dialog when openSCAD is full-screened. Instead, it creates a new tab which:

- confuses the purpose for a tab. It would make sense for multiple .scad files, but not for `n` .scad files and some Preferences tab.
- limits the user from seeing changed features. This requires a person to go back and forth
 
This is normal functionality, which works as expected.
[image]

This is what happens when the program is full-screened
[image]

This is what the terminal does when it is full-screened (expected behavior)
[image]

To bring up another issue with the preferences tab, I don't see why it should be a resizable dialog (or be able to be full-screened).

kintel, 2018-03-18:
Since we use Qt for GUI, it would be interesting if similar issues exist on other platforms.

MichaelAtOz, 2018-03-18:
> I don't see why it should be a resizable dialog

I do, it does not open to an optimal size, and you may not have seen the features tab. The Editor tab may well grow too. While I'm switching to 4K screen, not everyone has high resolutions.

> (or be able to be full-screened).

Comes with the territory, but it shouldn't to the tab thing above as a default.

t-paul, 2018-03-18:
I don't see a way to do actual full-screen on Linux.

closed 2026-03-05 by github-actions[bot]