openscad/openscad #2348: floor() function not working correctly

hech92, 2018-03-21:
Hello,

I don't know if this is a bug or because of some internal workings, but I have found a situation where floor() gives the wrong result:

`test=[for(i=[0:0.2:720]) [i,floor((i+0.0000000001)/360)] ];
echo(test[1799],test[1800],test[1801]);

test2=[for(i=[0:0.2:720]) [i,floor(i/360)] ];
echo(test2[1799],test2[1800],test2[1801]);`

When i equals 360, floor(i/360) should return 1, however it returns 0 (this is the case for "test2").
If I add a very small amount to i, floor works correctly and returns 1 ("test" case).

Tested with openscad 2015.03-2 and 2018.01.06, this issue exists in both versions.
OS is Windows 7 Professional 64-bit.

hech92, 2018-03-21:
After some further testing I'm not sure if the problem is the floor function or rather internal workings regarding floating point numbers. See code below for issues.

```
test = [ for(i=[600:0.5:603]) [i] ]; // counts to 603 (correct)
    echo(test);
    echo("test end --------------------------------------------");
test2 = [ for(i=[600:0.2:603]) [i] ]; // counts to 602.8 (wrong)
    echo(test2);
    echo("test2 end --------------------------------------------");
test3 = [ for(i=[600:0.2:602.8]) [i] ]; // counts to 602.6 (wrong)
    echo(test3);
    echo("test3 end --------------------------------------------");
test4 = [ for(i=[0:0.2:603]) [i] ]; // counts to 602.8 (wrong)
    echo(test4);
    echo("test4 end --------------------------------------------");
test5 = [ for(i=[0:0.2:602.6]) [i] ]; // counts to 602.6 (correct); if 602.6 in for loop is replaced by 602.8, this loop still counts to 602.6 (see test7), this seems to be a kind of "threshold" (values below 602.6 are always included, values above 602.6 are not included and the array ends 0.2 "early")
    echo(test5);
    echo("test5 end --------------------------------------------");
test6 = [ for(i=[600:0.2:602.6]) [i] ]; // counts to 602.4 (wrong); comparison with test5 above shows that end value is not all that matters
    echo(test6);
    echo("test6 end --------------------------------------------");
test7 = [ for(i=[0:0.2:602.8]) [i] ]; // counts to 602.6 (wrong)
    echo(test7);
    echo("test7 end --------------------------------------------");
test8 = [ for(i=[600:0.2:603+0.000001]) [i] ]; // counts to 603 (correct)
    echo(test8);
```

t-paul, 2018-03-21:
All numbers in OpenSCAD are floating point, 0.2 is not possible to encode in base2 floating point, so it's currently producing rounding issues. There are some ideas how to improve the special case with `for()` but the general rules for working with floating point values apply.

closed 2018-03-21 by t-paul