openscad/openscad #2349: [Feature Request] nested /* ... */ comments

NateTG, 2018-03-24:
Would make the /*  ... */ style comments a bit easier to use.

thehans, 2018-03-24:
I don't believe I've seen *any* programming languages treat block comments in this way, and would make OpenSCAD an outlier of abnormal comment handling as opposed to the way people are accustomed to.

NateTG, 2018-03-24:
I guess I should just use the editor for block commenting.

closed 2018-03-24 by NateTG

MichaelAtOz, 2018-03-25:
@NateTG I sympathise, that hits me so often, it's a wonder it has not been fixed over the decades in those other languages, but I agree with @thehans. (suggest closing this?)
When my new keypad arrives, I'm going to have '/*' & '*/' keys...

MichaelAtOz, 2018-03-25:
Markup removed the asterisks...`/*` & `*/` keys.