openscad/openscad #2351: Rendering Failure

NateTG, 2018-03-24:
These two render individually, but not together.

polyhedron(
   points=[[35.4769, 40.8432, -1], [34.6935, 41.958, -1], [33.7665, 42.9723, -1], [35.4769, 40.8432, 43.6585], [34.6935, 41.958, 43.6585], [33.7665, 42.9723, 43.6585], [25.1383, 34.2861, 11.2289], [26.2398, 35.1705, 9.8192], [25.1383, 34.2861, 43.6585], [26.2398, 35.1705, 43.6585]],
   faces=[[1, 0, 3, 4], [2, 1, 4, 5], [0, 1, 7, 6], [1, 2, 7], [2, 5, 9], [9, 7, 2], [7, 9, 8], [8, 6, 7], [6, 8, 3], [3, 0, 6], [5, 4, 3, 8, 9]]
);

polyhedron(
   points=[[34.6935, 41.958, -1], [33.7665, 42.9723, -1], [32.7121, 43.8687, -1], [34.6935, 41.958, 43.6585], [33.7665, 42.9723, 43.6585], [32.7121, 43.8687, 43.6585], [26.2398, 35.1705, 9.8192], [25.9674, 34.8881, 10.2108], [26.2398, 35.1705, 43.6585], [25.9674, 34.8881, 43.6585]],
faces=[[1, 0, 3, 4], [2, 1, 4, 5], [0, 1, 7, 6], [1, 2, 7], [2, 5, 9], [9, 7, 2], [7, 9, 8], [8, 6, 7], [6, 8, 3], [6, 3, 0], [5, 4, 3, 8, 9]]);

Console log:

PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1149 

Nota Bene: The cache behavior is pretty annoying when trying to reproduce console error messages.

Build info:

OpenSCAD Version: 2018.01.22
System information: Linux 4.9.0-6-686-pae #1 SMP Debian 4.9.82-1+deb9u3 (2018-03-02) i686 Debian GNU/Linux buster/sid 8 CPUs 16777216.00 TB RAM
Compiler: GCC "6.3.0 20170516" 32bit
Boost version: 1_62
Eigen version: 3.3.2
CGAL version, kernels: 4.9, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.7.1
QScintilla version: 2.9.3
MingW build: No
GLib version: 2.50.3
libzip version: 
Application Path: /usr/local/bin
Documents Path: /home/nate/.local/share
Resource Path: /usr/local/share/openscad
User Library Path: /home/nate/.local/share/OpenSCAD/libraries
User Config Path: /home/nate/.config/OpenSCAD
Backup Path: /home/nate/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/nate/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

NateTG, 2018-03-24:
Here's a version with triangular facets.

polyhedron(
   points=[
      [35.4769, 40.8432, -1],
      [34.6935, 41.958, -1],
      [33.7665, 42.9723, -1],
      [35.4769, 40.8432, 43.6585],
      [34.6935, 41.958, 43.6585],
      [33.7665, 42.9723, 43.6585],
      [25.1383, 34.2861, 11.2289],
      [26.2398, 35.1705, 9.8192],
      [25.1383, 34.2861, 43.6585],
      [26.2398, 35.1705, 43.6585]
   ],
   faces=[
      [1, 0, 3],
      [1,3, 4], 
      [2, 1, 4],
      [2,4, 5], 
      [0, 1, 7],
      [0,7, 6], 
      [1, 2, 7],
      [2, 5, 9],
      [9, 7, 2],
      [7, 9, 8], 
      [8, 6, 7],
      [6, 8, 3],
      [3, 0, 6],
      [5,4,3],
      [5,3,8],
      [5,9,8]
   ]
);

polyhedron(
   points=[
      [34.6935, 41.958, -1],
      [33.7665, 42.9723, -1],
      [32.7121, 43.8687, -1],
      [34.6935, 41.958, 43.6585],
      [33.7665, 42.9723, 43.6585],
      [32.7121, 43.8687, 43.6585], 
      [26.2398, 35.1705, 9.8192],
      [25.9674, 34.8881, 10.2108],
      [26.2398, 35.1705, 43.6585], 
      [25.9674, 34.8881, 43.6585]
],
   faces=[
      [1, 0, 3],
      [1,3, 4],
      [2, 1, 4],
      [2,4, 5], 
      [0, 1, 7],
      [0,7, 6], 
      [1, 2, 7], 
      [2, 5, 9],
      [9, 7, 2], 
      [7, 9, 8],
      [8, 6, 7], 
      [6, 8, 3], 
      [3, 0, 6], 
      [5,4,3],
      [5,3,8],
      [5,8,9]
]
);

NateTG, 2018-03-25:
Here's another pair with manually triangulated facets.
```
polyhedron(
   points=[[32.7121, 43.8687, -1], [31.5376, 44.6445, -1], [32.7121, 43.8687, 40], [31.5376, 44.6445, 40], [25.4921, 34.2553, 11.0008], [25.4921, 34.2553, 40], [32.7121, 43.8687, -0.999999], [31.5376, 44.6445, -0.999999], [32.7121, 43.8687, 40], [31.5376, 44.6445, 40]],
faces=[[1, 0, 2], [1, 2, 3], [0, 6, 8], [0, 8, 2], [3, 9, 7], [3, 7, 1], [0, 1, 4], [6, 0, 4], [1, 7, 4], [6, 4, 5], [6, 5, 8], [4, 7, 9], [4, 9, 5], [9, 3, 2], [9, 2, 8], [9, 8, 5]]);

polyhedron(
   points=[[31.5376, 44.6445, -1], [30.2503, 45.2988, -1], [31.5376, 44.6445, 40], [30.2503, 45.2988, 40], [24.8848, 33.2117, 12.2063], [24.8848, 33.2117, 40], [31.5376, 44.6445, -0.999999], [30.2503, 45.2988, -0.999999], [31.5376, 44.6445, 40], [30.2503, 45.2988, 40]],
faces=[[1, 0, 2], [1, 2, 3], [0, 6, 8], [0, 8, 2], [3, 9, 7], [3, 7, 1], [0, 1, 4], [6, 0, 4], [1, 7, 4], [6, 4, 5], [6, 5, 8], [4, 7, 9], [4, 9, 5], [9, 3, 2], [9, 2, 8], [9, 8, 5]]);
```

thehans, 2018-03-25:
I've seen these sort of assertion violations before from CGAL in a few other issues.  I searched and found the following issues that are possibly the same / related cause.
#396
#802
#1184
#1525

NateTG, 2018-03-25:
For what it's worth, triangulating the polyhedra differently seems to have resolved this issue for me.

tturpin, 2019-11-23:
Hello,
I think I am hitting the same bug. The code for me is:
`for(a=[0,120]) rotate(a) import("leg.stl");
`
and it triggers the same assertion violation `itl != it->second.end()` in the same file (at line 1152 for me).
I found  a workaround by passing the STL file to Meshlab's "Merge close vertices" filter.

MichaelAtOz, 2019-11-23:
@tturpin That is a different issue. External STL files are often broken. It is not a bug in OpenSCAD, as you say you fixed the STL in MeshLab.

tturpin, 2019-11-23:
I'm not so sure: the STL fil was generated by... OpenScad (as the difference between another STL file generated by another tool that I don't know, and a rotate_extrude of some profile.
Could you suggest a way of assessing whether an STL file is valid or not ? When I search, I found some basic explanations, but I couldn't find a link to a tool which simply answers yes or no to that question.
Also, if a broken STL file causes a crash, to me that is still a bug (a different one, I agree). I think that consistency should always be checked when opening files, probably in a permissive way, with just a warning, but this would help to sort this out.

MichaelAtOz, 2019-11-23:
MeshLab is a free tool, but you need to know what to look for (beyond the scope of this issue), or Netfabb which I think the trial version turns into a basic version after the trial, you need the basic automatic tests and also manually test for self-intersections. 
STL generate from another STL suggests the original is broken.
OpenSCAD didn't crash, it provided you a CGAL error telling you (badly) that bits of the geometry are inconsistent. There is discussion on whether to test imported STLs at that point.
That is different to this issue, which relates to OpenSCAD generated geometry having problems, ie polyhedron.

MichaelAtOz, 2019-11-23:
But finding the problem of broken STLs is an art that takes experience, fixing them more so.

tturpin, 2019-11-23:
I will see if Netfabb can run on wine... I must say that I don't understand why validating STL files (not fixing) should be an "art". Checking adjacent triangles orientations, holes, intersecting faces... does not seem extremely difficult. Even if there are different "degrees" of well-formedness, and degenerated geometry appearing from simple operations (like this example https://github.com/openscad/openscad/issues/396#issuecomment-19492027), I believe that serious checking is very welcome !

MichaelAtOz, 2019-11-23:
The Art is in understanding the tools.

> degenerated geometry appearing from simple operations 

Is mathematics. But this issue is not the place for these discussions.
Please use the Forum if you wish to discuss esoterics further.

tturpin, 2019-11-24:
@MichaelAtOz 
Sorry for polluting your 1.5 year old issue with my "esoteric" comments, but in fact you are wrong, I just reproduced the issue with the polyhedron function instead of import (using a simple sed script). So I end up with the same issue as was initially described (though with much larger polyhedra), no warnings and a core dump on assert failure. I know that it is redundant, but as I was not seeing any new comments during all this time, since the bug is serious, I just thought that it would be legitimate to mention that someone else faced it too, as it may indicate that it is not rare. I won't write here again.

tturpin, 2019-11-28:
Finally, I would like to add something. I found a reduced test case which basically reproduces the facet [0, 1, 7, 6] in second polyhedron of the original example:
```
polyhedron(points=[[1,0,0],[0,0,0],[-0.1,1,0],[0,2,0]],faces=[[0,1,3,2]]);

cube([1,1,1]);
```

With version 2019.11.19.nightly (git eedfcc4), I get
```
Expression : itl != it->second.end()
File       : /usr/include/CGAL/Nef_3/SNC_external_structure.h
Line       : 1152
```
When rendering with F6, but a different error when trying to output a .stl in command line:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: target(s1).vertex()==target(s2).vertex()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 227
```
Despite this difference, I think it's reasonable to assume that the crash in both the original example and this example are related somehow to the incorrect facet crossing itself.

MichaelAtOz, 2019-11-28:
@tturpin this is not a 'crash', OpenSCAD did not crash.
OpenSCAD output the [Assertion](https://en.wikipedia.org/wiki/Assertion_%28software_development%29) violation error message(s) that CGAL detected.

You have a malformed polyhedron, CGAL says 'you have a malformed polyhedron', but it is a bit technical in how it says it.

In the first case `itl != it->second.end()` [CGAL says](https://github.com/CGAL/cgal/blob/b9e320659e41c255d82642d03739150779f19575/Nef_3/include/CGAL/Nef_3/SNC_external_structure.h#L1144) one halfedge is not the same as the opposite halfedge.

In the second `target(s1).vertex()==target(s2).vertex()` [CGAL says](https://github.com/CGAL/cgal/blob/d6aabc311a2ed9be9c596e2c422687cfde335b99/Nef_3/include/CGAL/Nef_3/SNC_FM_decorator.h#L219) the two vertices to be intersected should be the same, but they are not.

> I think it's reasonable to assume that the crash in both the original example and this example are related somehow to the incorrect facet crossing itself.

Yes.
As Nate said
> For what it's worth, triangulating the polyhedra differently seems to have resolved this issue for me.

CGAL will not fix bad geometry any time soon. There is some discussion on whether to expend resources checking validity of poly's & `import` individually prior to render, but this could consume a lot of energy for no gain with correct geometry.

Thank you for reproducing the initial problem, confirming it is a malformed polyhedron.
This issue is now resolved. Closing.

closed 2019-11-28 by MichaelAtOz

MichaelAtOz, 2019-11-30:
@tturpin OpenSCAD uses a range of libraries to do the hard stuff.
So it has to live within the limitations of these libraries.

One of these limitations is that CGAL requires proper mathematically correct geometry, CGAL is full of assertions checking a vast range of requirements, eg in the two links above. 
I don't believe these a listed on one spot, so someone would have to go hunting to find out all the particular requirements. Or they would need to go hunting for other test mechanisms. That would be a precursor for any programmer to try to check them separately, and that programmer would need to have detailed knowledge the internal structures CGAL uses before trying to reproduce those checks in OpenSCAD code. They would also want to do that. That someone is not here ATM.

These assertions are deep in the guts so there is no clean way to handle them, basically it trips an exception, OpenSCAD catches it and is able to report the CGAL error message. It has no idea where in the geometry it occurred. There MAY be further mechanism within CGAL but someone would need to drill into it, and I'm just guessing but it would probably require restructuring OpenSCAD to do it a different way. 

You can get a feel for the complexity of the beast [here](https://doc.cgal.org/latest/Manual/packages.html).

Developers are aware that malformed poly's and bad imported geometry are a PITA, particularly for beginners. This is in the back of our minds should opportunities arise to help, but there a few developers and the cogs turn slowly.

MichaelAtOz, 2019-11-30:
@tturpin I suspect you have a different problem. You keep referring to crash and on the forum you say 'Aborted (core dumped)'. That does not match the behaviour I see, either GUI or command-line on Windows or Debian. 
The GUI should not crash with a CGAL assertion. Is that happening to you?
The command line should exit cleanly.

tturpin, 2019-11-30:
I'm running version 2019.11.19.nightly (git eedfcc4) on Ubuntu. The error message is the one when rendering with F6 from the GUI ; for the command line I get (suprisingly) a distinct error:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: target(s1).vertex()==target(s2).vertex()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 227

but the exit code is 0 and an STL is created, with the cube.

If ou don't reproduce the "windows closed" part with this nightly build, I can try to provide more details (maybe I changed some setting).

tturpin, 2019-11-30:
> [...] CGAL is full of assertions checking a vast range of requirements, eg in the two links above.
> I don't believe these are listed on one spot, so someone would have to go hunting to find out all the particular requirements. Or they would need to go hunting for other test mechanisms. [...]
I did not want to imply that a fully "safe" OpenScad is an easy and quick task to implement. Especially if assumptions are not explicitely and precisely specified (I trust you on this, I don't know the library). But I believe that the someone with the knowledge required to implement OpenScad on top of this library should be able to improve things significantly with, say, some days of work. From the two places in CGAL that you pointed with consistency checks, I cannot conclude how difficult it would be to achieve complete validation, and I don't know if any such checking code could be called explicitly, or if the best solution has to be outside CGAL. About catching CGAL exceptions, if OpenSCAD has no idea where it occurs in the geometry, then the handler is just not close enough to the call CGAL...