openscad/openscad #2352: Excessive calls to QGLView::paintGL on window focus

thehans, 2018-03-25:
I've noticed this a few times now when attempting to debug OpenSCAD with breakpoints.  It seems that focusing on the OpenSCAD window causes the  came across this issue while trying to debug some unrelated code.

steps to reproduce on Linux:
* build openscad with debug info enabled
* "gdb ./openscad"
* start the executable with "run" from gdb's command line
* After OpenSCAD window is loaded, move/tile it so that gdb console and openscad window are both on top and not obscuring each other
* Ctrl-C from gdb to interrupt, set breakpoint to start of QGLView::paintGL "b QGLView.cc::179" (line # may differ depending on git revision)
* mouse over the OpenSCAD *editor* pane and click to focus.
* the paintGL breakpoint will trigger, so go back to gdb, and use "continue" command
* it requires 4 continue commands before it stops hitting the breakpoint

Since the camera is not being moved, geometry hasn't changed, window is not resized, etc I would not expect paintGL to be called at all, much less 4 times.

On Linux Mint 18.3, my OpenSCAD is built against Qt 5.5.1

This isn't a noticeable effect normally, but I have had some cases where say $fn was accidentally set too high on a complex model, and preview is excessively slow taking a couple seconds per frame, it makes it really difficult to focus on the editor and just change the code back to something reasonable without a force quit.  And if you try to switch to other programs while you wait for OpenSCAD to do its thing, it just makes things worse because whenever you refocus its going to stall out again.

This issue also raises the question: Does it repaint *every* frame 4 times unnecessarily?

kintel, 2018-03-25:
It feels like this is smth. that was fixed by using `QOpenGLWidget`. Did you on purpose downgrade to `QGLWidget` ?

t-paul, 2018-03-25:
With the default `QOpenGLWidget` on Debian/Testing, I can't see any `paintGL()` call when switching windows. I did not try with debugger though.

thehans, 2018-03-25:
No, not on purpose.  Not sure what determines which one gets used.  

I have been building with cmake so maybe that's the difference?

I'm trying again with qmake now, but I'm getting an error that it can't find QAction
```
In file included from src/Preferences.h:7:0,
                 from src/scintillaeditor.cpp:7:
./ui_Preferences.h:13:29: fatal error: QtWidgets/QAction: No such file or directory
compilation terminated.
```
Not sure if its picking up some mix of Qt4 and Qt5 includes?

kintel, 2018-03-25:
The cmake build system should also enable QOpenGLWidget: https://github.com/openscad/openscad/blob/master/CMakeLists.txt#L127

..unless it's indeed picking up a mix of Qt versions..

thehans, 2018-03-25:
OK, I see that QGLView *inherits* QOpenGLWidget, so they are not mutually exclusive.
https://github.com/openscad/openscad/blob/master/src/QGLView.h#L20

It *does* appear to be using QOpenGLWidget, which can be seen in the full backtrace
```
#0  0x0000000000907cfd in QGLView::paintGL() (this=0xfd0700) at /home/hans/projects/openscad/src/QGLView.cc:179
#1  0x00007ffff40ebded in QOpenGLWidgetPrivate::invokeUserPaint() (this=0x10965b0) at kernel/qopenglwidget.cpp:812
#2  0x00007ffff40ccfc8 in QWidget::event(QEvent*) (this=0xfd0700, event=0x7fffffffb050) at kernel/qwidget.cpp:9044
#3  0x00007ffff408a05c in QApplicationPrivate::notify_helper(QObject*, QEvent*) (this=this@entry=0xd842c0, receiver=receiver@entry=0xfd0700, e=e@entry=0x7fffffffb050) at kernel/qapplication.cpp:3716
#4  0x00007ffff408f516 in QApplication::notify(QObject*, QEvent*) (this=0x7fffffffcb80, receiver=0xfd0700, e=0x7fffffffb050) at kernel/qapplication.cpp:3499
#5  0x00000000008f0e0f in OpenSCADApp::notify(QObject*, QEvent*) (this=<optimized out>, object=<optimized out>, event=<optimized out>) at /home/hans/projects/openscad/src/OpenSCADApp.cc:31
#6  0x00007ffff379738b in QCoreApplication::notifyInternal(QObject*, QEvent*) (this=0x7fffffffcb80, receiver=receiver@entry=0xfd0700, event=event@entry=0x7fffffffb050) at kernel/qcoreapplication.cpp:965
#7  0x00007ffff40c5ab9 in QWidgetPrivate::sendPaintEvent(QRegion const&) (event=0x7fffffffb050, receiver=<optimized out>) at ../../include/QtCore/../../src/corelib/kernel/qcoreapplication.h:227
#8  0x00007ffff40c5ab9 in QWidgetPrivate::sendPaintEvent(QRegion const&) (this=this@entry=0x10965b0, toBePainted=...) at kernel/qwidget.cpp:5625
#9  0x00007ffff40c6101 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0x10965b0, pdev=pdev@entry=0x1909210, rgn=..., offset=..., flags=flags@entry=4, sharedPainter=sharedPainter@entry=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5565
#10 0x00007ffff40c6dac in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfcbb70, pdev=pdev@entry=0x1909210, siblings=..., index=<optimized out>, index@entry=2, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5760
#11 0x00007ffff40c6c99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfcbb70, pdev=pdev@entry=0x1909210, siblings=..., index=2, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5747
#12 0x00007ffff40c5c71 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfcbb70, pdev=pdev@entry=0x1909210, rgn=..., offset=..., flags=flags@entry=4, sharedPainter=sharedPainter@entry=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5617
#13 0x00007ffff40c6dac in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfcb670, pdev=pdev@entry=0x1909210, siblings=..., index=<optimized out>, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5760
#14 0x00007ffff40c5c71 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfcb670, pdev=pdev@entry=0x1909210, rgn=..., offset=..., flags=flags@entry=4, sharedPainter=sharedPainter@entry=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5617
#15 0x00007ffff40c6dac in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfa1780, pdev=pdev@entry=0x1909210, siblings=..., index=<optimized out>, index@entry=82, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5760
#16 0x00007ffff40c6c99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfa1780, pdev=pdev@entry=0x1909210, siblings=..., index=82, index@entry=83, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5747
#17 0x00007ffff40c6c99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfa1780, pdev=pdev@entry=0x1909210, siblings=..., index=83, index@entry=84, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5747
#18 0x00007ffff40c6c99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfa1780, pdev=pdev@entry=0x1909210, siblings=..., index=84, index@entry=85, rgn=..., offset=..., flags=4, sharedPainter=0x0, backingStore=0x1427b80) at kernel/qwidget.cpp:5747
#19 0x00007ffff40c6c99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) (this=this@entry=0xfa1780, pdev=pdev@entry=0x1909210, siblings=..., index [... cut ...]
#43 0x00000000005e2e52 in gui(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, boost::filesystem::path const&, int, char**) (inputFiles=std::vector of length 1, capacity 1 = {...}, original_path=..., argc=argc@entry=1, argv=argv@entry=0x7fffffffdc38) at /home/hans/projects/openscad/src/openscad.cc:774
#44 0x00000000005d0bda in main(int, char**) (argc=1, argv=0x7fffffffdc38) at /home/hans/projects/openscad/src/openscad.cc:984
```

kintel, 2018-03-25:
Ah, right, I somehow read `QGLWidget`.
This could be a Qt-5.5 bug as well. I seem to remember there being some updates a while back related to this.

thehans, 2018-03-25:
Sorry I may have cause some confusion in my original post which I edited a couple times already.
I did type `QGLWidget` in one place where I meant `QGLView`

thehans, 2018-03-25:
I tried to understand the why it can't find `QtWidgets/QAction` by looking in the docs but annoyingly the Qt 5.5 online docs have been removed.  I guess its reached end of life according to Qt, which really sucks for Ubuntu LTS 16.04 that will be "supported" by Ubuntu until April 2021.

kintel, 2018-03-25:
To double-check, you can inspect `ui_Preferences.h` and see what version of Qt was used to generate that file, in case there is a discrepancy there.

thehans, 2018-03-25:
OK, I figured out the qmake issue.  The build was calling `uic` from qt4 for some reason.
I ran `which uic` and found `usr/bin/uic` which is a symlink to `/usr/bin/qtchooser`
So I saw from qtchooser that I can set env variable QT_SELECT=5 to force it.

With QT_SELECT set properly, I can complete the qmake build successfully, but it exhibits the same issue as the cmake build; painting 4 times per window focus.

thehans, 2018-03-25:
@t-paul how are you checking if it called paintGL without a debugger?

thehans, 2018-03-25:
A little more testing on my end, I added a PRINTDB statement to count calls to `QGLView::paintGL` as opposed to breakpoints, and I can now see that it gets painted twice on focus, and twice on defocus.

t-paul, 2018-03-25:
I did put in a `printf()` to output the current timestamp :-).

thehans, 2018-03-25:
I am watching the frame counter in console while trying various things aside from application window focus/defocus events (each of which paint 2 times).  

The View toolbar buttons all increment by a single frame when clicked, which is good.
Zooming with a single click of the mousewheel increments a single frame also.

However, if I orbit or pan the camera with click-drag of the mouse, moving very carefully by a single pixel, it seems that it always draws 2 frames at a time.  So I think some redundancy is happening there as well.

For reference I commited this small logging change that I'm using to a new branch:
https://github.com/thehans/openscad/commit/b07e793a37667901e9c03b84349ef3e86b55d361

thehans, 2018-03-25:
@t-paul what Qt version is your Debian/Testing build on?

t-paul, 2018-03-25:
```
System information: Linux 4.14.0-3-amd64 #1 SMP Debian 4.14.17-1 (2018-02-14) x86_64 Debian GNU/Linux buster/sid 8 CPUs 15.39 GB RAM
Compiler: GCC "7.3.0" 64bit
Boost version: 1_62
Eigen version: 3.3.4
CGAL version, kernels: 4.11.1, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.2
QScintilla version: 2.10.2
GLib version: 2.56.0
libzip version: 1.1.2

GLEW version: 2.0.0
OpenGL Version: 3.0 Mesa 17.3.6
GL Renderer: Mesa DRI Intel(R) UHD Graphics 620 (Kabylake GT2) 

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)
```

MichaelAtOz, 2018-03-25:
For what it's worth, I have suspected unnecessary repaints, on Windows, for some time, from released version thru snapshots, so I don't think it is new.  IIRC while in preview is significantly worse than F6.

If you have a slightly older congested system, and complex models, it tends to highlight the problem, where eg dragging the pane separate is practically impossible, while you can count refreshes as it limps across the screen.

Consider doing something to reduce repetitive repainting within a given small time period? Or an option to only repaint on release?

MichaelAtOz, 2018-03-26:
To explain further, while something else may be happening at such times, it appears as if the refreshes get queued, you can release the mouse, and still watch the window changes as if it is animated over some seconds.

thehans, 2018-03-27:
@MichaelAtOz Good point about the queuing phenomemon.  The more I think about it, the fact that its even possible for the paint calls to get queued up is the core issue here.

I worked on some changes so that the actual rendering happens through a QTimer which runs up to 60fps.  As I understand, if the slot for QTimer takes multiple intervals to complete, it will not queue up and only process the latest one.

thehans, 2018-03-27:
I'm also thinking maybe some kind of event filtering would be beneficial but that seems tricky because `QGLView::paintGL` calls are all caused by paint events, but those paint events are indirectly triggered by other events somewhere else.

I'm not particularly versed in Qt so any other suggestions on optimal frame rendering strategies are welcome.

henwist, 2018-09-06:
I have used another technique in a program I wrote to zoom in/out with the scroll wheel. I just used one of the events for the scroll wheel to zoom in and then discarded the other enqueued events of one type. This led to a very responsive user interface compared with the former, where many scroll wheel events needed to be processed and that took some extra time per scroll wheel move.

thehans, 2019-02-06:
I've since updated to Linux Mint 19, and my builds are now against Qt 5.9.5, with which I can no longer reproduce any extra frame draws.

closed 2019-02-06 by thehans