openscad/openscad #2357: Segmentation fault (core dumped)

romanw, 2018-03-28:
The following code causes openscad (both git and release versions) to segfault...

```
$fn=96;

difference() {
cube([12, 12, 2], center=true);
cube([8, 8, 2], center=true);
}
```

Regardless of the correctness of the above code, openscad should handle it gracefully and not segfault.

romanw, 2018-03-28:
Seems to be opengl related. Can workaround it with...

```
LIBGL_ALWAYS_SOFTWARE=1 openscad
```

Alternatively, as per the FAQ, `Edit->Preferences->Advanced->Force Goldfeather`, appears to improve the situation.

t-paul, 2018-03-28:
Sounds like a duplicate of #2041. Are we talking about Windows? Which OpenSCAD version? If yes, please share the info from the Event Viewer.

romanw, 2018-03-31:
Linux 4.15 running on an old HP 6715s laptop which has...
```
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RS690M [Radeon Xpress 1200/1250/1270]
```
Graphics driver is ```xf86-video-ati 1:18.0.1```

Output of ```glxinfo``` in attached file --> [glxinfo.txt](https://github.com/openscad/openscad/files/1865723/glxinfo.txt)

OpenSCAD versions 2015.03.03 and 20180324.

t-paul, 2018-04-01:
1) What distribution is that? Linux 4.15 sounds just like the kernel version?
2) Are those the builds downloaded from the main web page and OBS as linked there?
3) Please post the output `Help->LibraryInfo`
4) Do you know how to get a backtrace of the crash? `gdb openscad` and such?

t-paul, 2018-04-01:
@MichaelAtOz Good idea, maybe just link this [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?) entry? That way we need to update just one place if we want to improve the text.

thehans, 2018-04-01:
@t-paul @MichaelAtOz your recent comments make me think some github guidelines etc would be helpful, so I've created issue #2364 for some further discussion.

romanw, 2018-04-05:
Regarding the questions...
1. The distribution is ArchLinux
2. I initially used the OpenSCAD version (2015.03.03) provided by ArchLinux (https://www.archlinux.org/packages/community/x86_64/openscad/). After that I tried the git version from the ArchLinux User Repository (https://aur.archlinux.org/packages/openscad-git).
3.  Output from `Help->LibraryInfo` ...
```
OpenSCAD Version: 20180324
System information: Linux 4.15.14-1-zen #1 ZEN SMP PREEMPT Wed Mar 28 17:43:56 UTC 2018 x86_64 Arch Linux 2 CPUs 3.73 GB RAM
Compiler: GCC "7.3.1 20180312" 64bit
Boost version: 1_66
Eigen version: 3.3.4
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.10.1
QScintilla version: 2.10.3
MingW build: No
GLib version: 2.56.0
libzip version: 
Application Path: /usr/bin
Documents Path: /home/user/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/user/.local/share/OpenSCAD/libraries
User Config Path: /home/user/.config/OpenSCAD
Backup Path: /home/user/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/user/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 2.1.0
OpenGL Version: 2.1 Mesa 17.3.7
GL Renderer: ATI RS690
GL Vendor: X.Org R300 Project
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_copy_buffer
GL_ARB_debug_output
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_objects
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_MESA_ycbcr_texture
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```
4. I'll try to get a backtrace and post in a seperate comment.

I guess I should have RTFW before opening this issue. Sorry.

romanw, 2018-04-05:
Backtrace of the crash...

```
$ gdb openscad
GNU gdb (GDB) 8.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from openscad...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/openscad 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffe97c2700 (LWP 10191)]
qt5ct: using qt5ct plugin
[New Thread 0x7fffe2f85700 (LWP 10192)]
[New Thread 0x7fffd71d5700 (LWP 10193)]
qt5ct: custom style sheet is disabled
qt5ct: D-Bus global menu: no

Thread 1 "openscad" received signal SIGSEGV, Segmentation fault.
0x00007fffe8035774 in ?? ()
(gdb) bt
#0  0x00007fffe8035774 in ?? ()
#1  0x3f8000003f800000 in ?? ()
#2  0x3f8000003f800000 in ?? ()
#3  0x0000000000000000 in ?? ()
(gdb) 
```

henwist, 2018-09-03:
I have looked deeper into the code regarding this segfault. I found that it is the ATI driver for old (2006 in my case) graphics cards in combination with mesa that triggers this error.

**Library info from OpenSCAD:**
OpenSCAD Version: 2015.03-3
System information: Linux 4.17.7-200.fc28.x86_64 #1 SMP Tue Jul 17 16:28:31 UTC 2018 x86_64 Fedora 28 (Twenty Eight) 1 CPU 3.60 GB RAM
Compiler: GCC "8.0.1 20180324 (Red Hat 8.0.1-0.20)" 64bit
Boost version: 1_66
Eigen version: 3.3.4
CGAL version, kernels: 4.11.1, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 4.8.7
QScintilla version: 2.10.2
MingW build: No
GLib version: 2.56.1
Application Path: /usr/bin
Documents Path: /home/liveuser/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/liveuser/.local/share/OpenSCAD/libraries
User Config Path: /home/liveuser/.config/OpenSCAD
Backup Path: /home/liveuser/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/liveuser/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 2.0.0
OpenGL Version: 2.1 Mesa 18.0.5
GL Renderer: ATI RS480
GL Vendor: X.Org R300 Project
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_copy_buffer
GL_ARB_debug_output
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
[... 335 lines cut ...]
         pipeline( fpme, vert_info, prim_info );
      }
      else {
         emit( fpme->emit, vert_info, prim_info );
      }
   }
   FREE(vert_info->verts);
   if (free_prim_info) {
      FREE(prim_info->primitive_lengths);
   }
}`

**This line will segfault deep deep inside mesa library:**
/usr/src/debug/mesa-18.0.5-3.fc28.x86_64/src/gallium/auxiliary/draw/draw_pt_fetch_shade_pipeline_llvm.c:400:   clipped = fpme->current_variant->jit_func(&fpme->llvm->jit_context, ...

I cannot see how to trace inside jit_func , but somebody may seek deeper in mesa... anyway this segfault seems to be in mesa so older hardware with ATI drivers combined with mesa doesn't work - e.g: 
OpenGL Version: 2.1 Mesa 18.0.5
GL Renderer: ATI RS480
GL Vendor: X.Org R300 Project. 

GL Renderer: llvmpipe with mesa and ATI works though.

**So a conclusion all in all as I see it:**  - the bug is most probably a mesa bug with X.Org R300 Project for the GL Renderer : ATI RSnnn.

/Henrik

donbright, 2018-09-03:
thanks @henwist 

i have seen this happen too... i think we should explore the idea of blacklist that auto-switches to software rendering for certain cards (on windows QT this can be done also with QT_ANGLE_PLATFORM=warp , on linux at the least LIBGL_ALWAYS_SOFTWARE=1 ) ..

nophead, 2018-09-03:
Perhaps a preference the user could change, like Force Goldfeather. That way it doesn't need a new addition to the blacklist each time a user comes across a problem. They can just flip the switch to see if software rendering fixes it.

donbright, 2018-09-03:
thats a good point however this type of error can crash OpenSCAD on startup so the user doesnt have a chance to see any switch they can flip, it just crashes without explanation.

nophead, 2018-09-03:
Perhaps an environment variable to force it on?

donbright, 2018-09-03:
a program should never crash on startup. 

this is why web browsers and other projects have 'blacklists' of GPUs where they force software rendering (the user can try to use the terrible drivers if they want by overriding the blacklist)

https://groups.google.com/a/chromium.org/forum/#!topic/chromium-dev/AQNCH7plJe4

https://github.com/mavlink/qgroundcontrol/issues/2776
 
https://stackoverflow.com/questions/34796504/what-happens-to-blacklisted-graphic-cards-in-qt-5-6-when-running-a-qml-app

kintel, 2018-12-07:
Closing as driver bug. If someone wants to play with blacklisting, we can open a feature request for that.

closed 2018-12-07 by kintel