openscad/openscad #2358: Baidu engines detects "Win32.Trojan.WisdomEyes.16070401." in OpenSCAD-2015.03-2-x86-64-Installer.exe

HaraldMuehlhoffCC, 2018-03-28:
Firstly thanks for a great tool! So far I've only installed OpenSCAD on a Hyper-V VM because the Baidu engine on virustotal.com detected the "Win32.Trojan.WisdomEyes.16070401." in OpenSCAD-2015.03-2-x86-64-Installer.exe download from http://www.openscad.org/downloads.html. My local scanners - Windows Defender on Windows 10 & Malwarebytes - didn't detect anything.

I tried to submit a false positive report to Baidu but the web interface didn't work (Edge & Chrome) (http://antivirus.baidu.com/en/submit-url.php).

Link to the Virustotal scan: https://www.virustotal.com/#/file/316b95c1fc029bdf493b752d9917cd895dbfb5ddf9a4561b27efb3728d3d74e4/detection

PS: I wonder why the Virustotal SandBox (Behavior tab) finds a System.dll file written to. 

```
Files written
C:\DOCUME~1\<USER>~1\LOCALS~1\Temp\nsb2.tmp\System.dll
```

kintel, 2018-03-28:
I deem this a false positive. Baidu will eventually get it right.
I guess that the System.dll is a side-effect of the scanner itself.

choller, 2018-03-29:
System.dll is part of your installer. From what I understand it is a generic NSIS plugin to run arbitrary code during installation. Do you use this plugin to do anything during installation?

kintel, 2018-03-29:
@choller Thx, good to know. I haven't looked too closely at the Windows installer, but that's all part of our sources: https://github.com/openscad/openscad/blob/master/scripts/installer.nsi

We do write to the registry. Perhaps some virus scanners don't like that.

donbright, 2018-08-04:
NSIS itself might be a problem. https://bugzilla.mozilla.org/show_bug.cgi?id=709738

https://www.scmagazine.com/microsoft-malicious-nsis-installers-used-to-disperse-ransomware-evolve-and-grow-in-number/article/644849/

we really should be using more modern system... and signing our code somehow

donbright, 2018-08-04:
step 1 eval alternatives
step 2 implement experimental x-build
step 3 test
step4 integrate w build scripts

- [ ]

t-paul, 2019-11-18:
Ok, closing this as it never was an actual infection and leaving that open for changing the installer looks confusing. I'll move the suggestion to a new issue with a clear topic line.

closed 2019-11-18 by t-paul