openscad/openscad #2360: Possible Improvement using CGAL "Polygon Mesh Processing" for booleans and more?

thehans, 2018-03-30:
OpenSCAD is currently using CGAL Nef Polyhedra for boolean operations, which can be quite slow for complex models.

I was looking at other options within CGAL and I found this section about mesh "corefinement" and booleans:
https://doc.cgal.org/latest/Polygon_mesh_processing/index.html#Coref_section

It sounds like this could be much faster than Nef processing, though maybe not as robust in some ways?
Although even if not intrinsically as robust, this also looks like it may be capable of better detecting and even possibly fixing mesh errors with the various functions provided (the [Hole Filling](https://doc.cgal.org/latest/Polygon_mesh_processing/index.html#PMPHoleFilling) section looks pretty impressive).

According to the package [overview page](https://doc.cgal.org/latest/Manual/packages.html#PkgPolygonMeshProcessingSummary), this was introduced in CGAL 4.7.  
OpenSCAD currently only requires CGAL 3.6, so this would be a large bump to our required version if this change were made.

kintel, 2018-03-31:
I did some early experimentation with this back in 2013 (see https://github.com/openscad/openscad/tree/epec-fast). The corefinement stuff appears to be slightly less robust than Nef3 polyhedrons, but it feels like something we may be able to work around.
The CGAL requirement was a bit of a showstopper back then, but this has been available for a while now, so perhaps time to give it another shot.

MichaelAtOz, 2018-03-31:
How about 4.8? [Multithreading???](https://doc.cgal.org/4.8/Manual/devman_multithreading.html)

kintel, 2018-03-31:
I haven't looked into whether corefinement is thread-safe, but since they use the same datatypes as Nef3 polyhedrons, my guess is that the same limitations apply.

thehans, 2018-04-03:
@kintel Was the epec-fast branch in a state that built and ran at any point or was it still WIP last time you left it?

I checked it out for a quick test but the build fails looking for `Dock.h`.  

Do you have some recommendations for next steps to move this forward?  I could give it a shot rebasing the branch to get it up to date with master?

kintel, 2018-04-04:
I cannot remember how I left it. It was really just a quick and dirty test, without any performance optimizations as I converted everything back and forth to Nef3 polyhedrons. My goal was to determine the CSG operation stability implications of using corefinement, but I had too many other things on my list..
The large missing piece to integrate properly is to better hide the interface to the underlying geometry engine classes. One of the challenges there is that caching may need a bit of a rewrite. See #1399 for an early external attempt at doing this.

dpellegr, 2020-09-30:
I think that the time is mature for this.

I used PyMesh to manipulate a pretty complex model which was failing/taking forever in OpenSCAD.
PyMesh allows to seamlessly switch between various rendering engines such as corefinement, nef3, igl, carve, cork.

I really pushed them with A LOT of complex differences operation. In my experience corefinement is now not only the fastest (by a couple of order of magnitude compared to nef3) but also the most stable and reliable.

ochafik, 2021-02-14:
I believe I've found a good balance of speed (10x to 100x in my little [grid benchmarks](https://gist.github.com/ochafik/2db96400e3c1f73558fcede990b8a355)) and robustness 
(avoiding cases that break Polygon Mesh Processing and reverting to Nefs transparently) in https://github.com/openscad/openscad/pull/3641

PR #4087 (2022-02-06): Experimental fast-csg feature: it’s time to render more complex models :-D
This is a cleaned up, minimal version of fast-csg (incubated in [PR#3641](https://github.com/openscad/openscad/pull/3641)), which greatly speeds up a wide range of models when the feature is enabled (in the GUI’s settings or with the `--feature=fast-csg` command-line flag), e.g. [these tiny benchmarks](https://gist.github.com/ochafik/2db96400e3c1f73558fcede990b8a355). Fixes #2360.

I've tried to itemize it in reasonable commits, but happy to send them as separate PRs instead.

Things I've left out for now:
- The fix to  [... cut]
closed 2022-02-06 by t-paul