openscad/openscad #2362: Misaligned characters and cursor in editor area (Windows10/IntelHD620)

vguenther, 2018-03-31:
When entering text in the Editor the cursor does not proceed the full character width.

[image]

When entering brackets, these are placed in the displayed cursor position rather than behind the last character.

[image]

When opening an scad-file, similar behaviour occurs.

[image]

I tried it with the Windows snapshots OpenSCAD-2018.03.10-x86-64-Installer.exe and OpenSCAD-2018.03.10-x86-64-Installer.exe for windows in Windows10. I am quite sure it was initially ok after the first installation of OpenSCAD-2018.03.10-x86-64-Installer. I observe the issue on one single Windws10 Notebook with Intel HD620 

[image]

On this notebook the problem is reproducible even after new installation of openSCAD. On another Windows10 Notebook everything is working fine.  (both on same -current- Windows version)

Behaviour occurs no matter if simple editor or qscintilla is selected.
It does not seem to depend on neither font nor size.
Nothing worth mentioning in the registry entries.

Any hints how to get around this?

Best regards, 
Volker

t-paul, 2018-03-31:
I'm not sure how that's possible except with a font that's either broken or not understood by the editor component we are using. Can you try changing the editor font in preferences and see if that issue goes away with a different font?
The default selection for the editor font sometimes seems to be a bit random.

vguenther, 2018-03-31:
Actually I tried many editor fonts in the preferences dialog. Every single one I tried showed the described behaviour. Changing the size did not help either.

vguenther, 2018-04-01:
I found out more about it. The issue occurs in a setup with an external monitor with a higher resolution.

The pattern is: 
Step 1) I open an scad-file on the internal display -> text in editor looks fine.
Step 2) I drag the window with the opened file to the external display -> text in editor still looks fine. But: it immediately messes up when I enter a single character.
Step 3) I close OpenScad and reopen it with doubleclick on filename, forcing it to reopen on external display -> text in editor looks messed up as shown in the original post.
Step 4) I drag the window to the internal display -> text in editor still messed. But after entering a character it looks fine again.
Step 5) I close and reopen on internal display - text in editor looks fine.

t-paul, 2018-04-01:
Ok, I guess we can't do muich about this then. We have no control over the internal rendering of the GUI framework. That sounds a bit like https://bugreports.qt.io/browse/QTBUG-48242 which is supposedly fixed in Qt 5.9 but OpenSCAD-2018.03.10 should already have that.

MichaelAtOz, 2018-04-01:
I'll try to remember to test this on my other PC, on W7, where - from memory, I set the screen DPI to a non-STD higher resolution, and I seem to recall similar issues. 
Also this DPI change, I suspect, makes things like the splash window, be non optimal size.
Unless you think the problem is elsewhere?

t-paul, 2018-04-01:
I see similar things like tiny window sizes on the new notebook which has 3840x2160 at 13". So running (on Linux) with 2 times scaling gives a nice sharp picture. So while the font is scaled ok for the welcome screen, the window itself is too tiny to look sensible, same with the preferences window.

Those things we can probably do something about. But the actual font rendering and especially that strange positioning issue comes very likely from either somewhere in Qt (supposedly fixed) or QScintilla (the text editor component, maybe ignoring that fix). I can't find a bug tracker for QScintilla, maybe the mailing list is the only option for reporting issues.

With (as far as I'm aware) no developer working on Windows there's not a good chance to dig that deep into third party code to actually fix that soon.

kintel, 2018-04-01:
This sounds like a duplicate of #1624 

There was one report of this symptom going away after switching to the simple editor (note that it requires a restart of OpenSCAD to take effect).

vguenther, 2018-04-01:
I just double-checked: Changing preferences from "QScintilla" to "simple editor" and restarting OpenSCAD does not make the symptom go away.

[image]

kintel, 2018-04-01:
@vguenther Thx, that definitely take QScintilla out of the loop. That also means that this should likely be reproducible with a plain Qt application.

t-paul, 2018-04-01:
I'm pretty sure that is not yet showing the simple editor. That has no folding, has it? That setting will need OpenSCAD to restart to be effective I think.

kintel, 2018-04-01:
Good point. @vguenther Could you try again?

vguenther, 2018-04-01:
I again checked the registry settings and restarted OpenSCAD again. Both registry and preferences say "simple editor", but symptom is still there. I agree there seems to be no visual difference at all. As a newbie I'm not aware of the differences between the editors, but to me it looks like OpenSCAD just ignores the editortype setting - which seems to an issue not directly related to the original one. 
I will try to find out which editor is actually used and I will have a closer look at #1624 asap and update accordingly.

t-paul, 2018-04-01:
Hmm, strange, I checked with both Linux version and Windows version via wine. In both cases switching does work. No idea why it's not working natively.

vguenther, 2018-04-01:
ok - I just learned folding is a discriminator and I definitely still have folding. I compared it to Linux version in virtualbox. I confirm switching works in Linux but not in Windows. Should I open another issue for that?

kintel, 2018-04-01:
Could the editor switching issue be related to language setting? I see we compare some strings against the English combobox contents, while the screenshot above show a German locale.

MichaelAtOz, 2018-04-02:
Switching works on my W7/64, with OpenSCAD 2018.01.06 (latest I have ATM).

t-paul, 2018-04-02:
Yep, it's picking up the translated value so that never matches. See #2365.

t-paul, 2018-04-07:
Updated Windows snapshots are online. Please try again with those. Switching the editor type should now work regardless of the language settings.

vguenther, 2018-04-08:
I confirm now editor switching works fine with German language setting. Thx. I will try in my dual screen setup asap to finally confirm this issue is a duplicate of #1624.

vguenther, 2018-04-08:
As expected the issue is specific to qscintilla editor. It goes away after switching to simple editor. So we obviously have a duplicate.

kintel, 2018-04-09:
@vguenther Thanks for following up! Closing this as a duplicate of #1624

closed 2018-04-09 by kintel