openscad/openscad #2363: Customizer Doesn't Work in Latest Win64 Installer Snapshot

TekMason, 2018-04-01:
OpenSCAD 2018.03.10 (git a77a775)
Windows 10 Pro 64 bit version 1709 build 16299.334
Customizer doesn't work with OpenSCAD-**2018.03.10**-x86-64-Installer.exe

Customizer had been performing without any issues using OpenSCAD 2018.01.06 (git 3473eb1) with the OpenSCAD-2018.01.06-x86-64-Installer.exe

After uninstalling 2018.01.06 and installing the OpenSCAD 2018.03.10 Win64 Customizer stops working.  The Customizer pane is active and available  but does not display any variables or an Expand Node Caret.

I have tried everything I can think of with the 2018.03.28 snapshot but nothing fixes it.
Uninstalling it and reinstalling the older 2018.01.06 build fixes it.

TM

kintel, 2018-04-01:
@MichaelPFrey Any ideas?

TekMason, 2018-04-01:
I tested with Dr. Lex's CustomizerExample.scad from Thingiverse
https://www.thingiverse.com/thing:2812634

I see that MichaelPFrey has made quite a few Customizer commits since 2018.03.10, unfortunately, I don't have the wherewithal to compile from source to test them :(.

MichaelPFrey, 2018-04-01:
fixed with #2338 Unfortunately, building for or on windows is tricky.

@t-paul would you mind making a new windows installer?

t-paul, 2018-04-01:
Windows Snapshots are updated. Customizer looks fine via wine on Linux.

TekMason, 2018-04-01:
Confirmed resolved on OpenSCAD 2018.04.01 (git e7c0851) from OpenSCAD-2018.04.01-x86-64-Installer.exe

After opening a file and pressing F5 or F6, the Customizer items become available and there is an Expand Caret with the text "Parameters" adjacent to it (image at end of post).  An improvement over the 2018.01.06 build there was just a bare Expand Caret.

Thanks for the rapid response on that guys and the new Windows Development Snapshots t-paul.
I will be sure to update my posts in other forums to announce the new build.  I'm sure you are all seeing a lot more Customizer related question since Thingiverse Customizer functionality has been broken.

A few minor add on questions that I will gladly create new issues for if you wish:
1. Should items in the Customizer pane be available after a file is loaded (before F5 or F6)?  In other builds I believe the first row of items were available.
2. Should a Preview (F5) be automatically executed after a new file is opened?
3. At http://www.openscad.org/downloads.html the items on the left under Downloads link to anchors for each item however the user isn't taken to the relevant section i.e. Development Snapshots when it is selected.
4. ~~Change the link under Downloads > Development Snapshots to point to the new builds.~~

[image]

TM

t-paul, 2018-04-01:
1) No, it requires parsing+preview of the file which is currently not done (unless see 2). I guess that would be a nice new feature to just do parsing without the preview.
2) Activating `Design->Auto Reload and Preview` does have that effect, it's mainly intended for reloading externally changed files though (e.g. due to using an external editor)
3) Works for me (Firefox)

thehans, 2018-04-01:
3. Has been a problem with Chrome for a while, not sure when they made the change but its really horrendous that they broke the majority of `#location` url on the internet.
I *think* the solution is that all links must point to an actual anchor `<a href="#blah"></a>`

kintel, 2018-04-01:
@thehans Do you have more info about that. I noticed the same but haven't investigated..

closed 2018-04-01 by kintel

thehans, 2018-04-01:
@kintel not really, mainly just my personal experience that many sites I use stopped working in this regard.
I did a bit of googling but haven't found any specifics on the issue.

One other possibility is that maybe its failing due to some invalid html, of which it seems there is a lot on the page:
https://validator.w3.org/nu/?doc=http%3A%2F%2Fwww.openscad.org%2Fdownloads.html