openscad/openscad #2364: Add Github-specific guidelines and templates

thehans, 2018-04-01:
This is a suggestion to add some files to OpenSCAD repo for integrating guidelines and templates with Github.

There are a few files that Github looks for (in root, `docs` or `.github` folders), and when they exist it uses these to prompt users with templates or links to guidelines etc.

These are 4 that I am aware of:
https://help.github.com/articles/setting-guidelines-for-repository-contributors/
https://help.github.com/articles/creating-an-issue-template-for-your-repository/
https://help.github.com/articles/creating-a-pull-request-template-for-your-repository/
https://help.github.com/articles/adding-a-code-of-conduct-to-your-project/

I'm not sure that all of them are completely necessary, but I think at least an Issue template would be useful towards improving the quality of reported issues.  Also in my opinion the `.github` folder would be the best path choice for these files.

Here is an example of a project which has these files in a `.github` folder:
https://github.com/MarlinFirmware/Marlin/tree/1.1.x/.github

We do have existing documentation in a few places, some of which might be useful to copy/paraphrase into potential new github-specific stuff.  In the wiki manual, in our `doc` folder (*not* `docs` that github looks for), and in pinned posts on forums.
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?
http://forum.openscad.org/Reporting-Bugs-not-just-general-queries-td21787.html

t-paul, 2018-04-01:
Yes, I eyed those a couple of times already. I'm not sure that helps a lot when just puting in the docs. Just look at the latest issues created on the Marlin repo. It does not seem to help. It might work together with some automatic enforcement like the [bot](https://github.com/foosel/GitIssueBot) Gina uses for Octoprint.

t-paul, 2023-04-17:
Issue templates added via #4607.