openscad/openscad #2369: Feature request: Font list on console

ospalh, 2018-04-03:
I would like a way for OpenSCAD to list the fonts when invoked from the command line. Maybe `openscad --fontlist`.
Recently, i wanted to change the font used in a project and then render it without the GUI, but, without X running on the Linux machine at the time, couldn’t get to the Help→Font List window and the exact name of the font and style.
Version used: 2015.03-3, on Fedora Linux, installed via package manager.

t-paul, 2018-04-03:
It's just using the system fonts managed by `fontconfig`. You can list all fonts with `fc-list`

closed 2018-04-03 by t-paul

ospalh, 2018-04-03:
Thanks.
But it’s not in the manual page.

t-paul, 2018-04-03:
Added note to wiki, this produces similar list as the GUI shows
```
fc-list -f "%-60{{%{family[0]}%{:style[0]=}}}%{file}\n" | sort
```