openscad/openscad #2370: Add option to downsample image imported using surface();

Harvie, 2018-04-03:
Hi!
I made surface from hi resolution heightmap. Like `surface("snezka.png",center=true);`. That results into high poly model. I want to reduce that so i can work with it and switch it back to full poly as soon as i'll go to export it.

I've resized the picture in kolourpaint, which kinda works. But it's extra work outside of openscad, it cannot be dynamicaly changed during openscad work (as i need to manualy resize, save and change path in openscad each time i want to). Also that changes XY size of resulting model, since surface() size is dependent on image size.

Idealy there would be something like this: `surface("snezka.png", details="21%");` that would smooth-scale eg. 200x200px image to 42x42px.

Is that possible?
Thanks in advance

t-paul, 2018-04-03:
It's possible, but also a huge can of worms as OpenSCAD is not intended to be an image manipulation framework. It still might be useful to have at least some limited control without the need of external programs.

Harvie, 2018-04-03:
Another way would be to generate hi-poly surface and then simplify it using quadratic mesh decimation, but that would be much slower. Maybe just calling external program like image magick to resize the image would be enough for now. But that adds another dependency, which is not very nice for such minor feature.